solmoller/eversolar-monitor #10: DBD::SQLite::db do failed: unrecognized token (eversolar.pl line 892)

StephenE, 2015-08-21:
I'm getting the following error every time the current usage is read from my invertor

DBD::SQLite::db do failed: unrecognized token: "'EL36806011540195" at eversolar.pl line 892.

The power is being displayed correctly, so I assume this is just a problem inserting the recorded values into the database.

solmoller, 2015-08-22:
Hi Stephen

1 Did you install from scratch, or use the SD image?
2 How big is the sqlite file?
3 if you increase the debug value, what does the log say?

It sounds like you are close to having a working system

StephenE, 2015-08-22:
To answer your questions:

1. Installed from scratch
2. 2kb
3. This is already at debug logging level 3

I altered the script to write out the SQL statement rather than executing it, and it looks like the following:

INSERT INTO inverter
                    (serial_number,
                     timestamp,
                     pac,
                     e_today,
                     e_total,
                     vpv,
                     vpv2,
                     ipv,
                     ipv2,
                     vac,
                     iac,
                     frequency,
                     impedance,
                     hours_up,
                     op_mode,
                     temp)
                VALUES
                    ('EL36806011540195',
                     '2015-08-22 08:10:44',
                     '803',
                     '1.33',
                     '197.6',
                     '181.2',
                     '181.9',
                     '2.3',
                     '2.3',
                     '239.7',
                     '3.3',
                     '49.97',
                     '262',
                     '250',
                     '1',
                     '26.2'
                    )

I don't immediately see a problem with that...

solmoller, 2015-08-23:
OK, 

1 did you clone from github og google? You would be the first from github, as I just  migrated a few days ago. What os are you on?
2 is anything written to the SQL file at all?
3 I'll look into my own debug and report back

StephenE, 2015-08-23:
1) I cloned from Google a couple of weeks back, but couldn't get the software working at the time because the USB to serial adapter I had wasn't the right type.
2) Just the table

Interestingly, when I hard coded the serial_number in the SQL statement to 'EL36806011540195', the error went away. Maybe there's something about the serial_number of our inverter that it doesn't like - this would explain why I was also having a problem getting it to upload to pvoutput (See issue #9).

solmoller, 2015-08-24:
ok, so it's not the cloning

Does it write to the SQL file now?

StephenE, 2015-08-25:
Yes, it writes to the table with the hard coded serial number

StephenE, 2015-08-27:
I undid my hard coding of the serial_number and everything worked fine. I then deleted the db.sqlite file and re-ran the script - at which point the problem came back. Looking at the code, I see that it uses a different path if it's trying to register an inverter for the first time compared compared to when there is an existing database.

As suggested, I printed out the length of the string and it seems to be longer than expected

    received packet from inverter:
    aa 55 00 00 01 00 10 80 12 45 4c 33 36 38 30 36 30
    31 31 35 34 30 31 39 35 00 01 05 05
    2015-08-27 19:27:45: Severity 1, Found serial number: EL36806011540195 18

It's two characters longer than the string returned previously. Here is the same log message printed out with the old db.sqlite file

    2015-08-27 19:21:58: Severity 2: got serial EL36806011540195 from database
    2015-08-27 19:21:58: Severity 1: Try to register known serial number: EL36806011540195 16
    sending packet to inverter...
    aa 55 01 00 00 00 10 01 11 45 4c 33 36 38 30 36 30
    31 31 35 34 30 31 39 35 10 04 94
    received packet from inverter:
    aa 55 00 10 01 00 10 81 01 06 01 a8

Sadly the sun has gone in now so I'm not able to get any further with this tonight. Any thoughts as to where those extra two characters are coming from?

StephenE, 2015-08-31:
Found what I believe to be the problem!

In the function register_inverter, where it unpacks the serial number response:

    my $serial_number = unpack("A*", $serial_num_response);

It's using the A* format specifier, which I believe is "null padded" string. I think this should be Z, which is "null terminated" string. This then gives me a serial number that is 16 characters long rather than the unexpected 18 that I've been seeing. I guess my inverter must have a shorter serial number than usual?

solmoller, 2015-08-31:
Hi Stephen
Nice finding!
I think we are on the home stretch now

If changing to null terminated string does not work, then I found this on perlmonks:
If you want to remove everything except alphanumeric characters and underscores, you can simplify it like this:

$str =~ s/\W//g;

That should take out funny characters

Henrik

StephenE, 2015-09-02:
Z* has fixed the issue for me

solmoller, 2015-09-02:
ok I have updated the code and closes this one

closed 2015-09-02 by solmoller