solmoller/eversolar-monitor #11: Perl occasionally exits early in the morning

StephenE, 2015-09-02:
I've seen this twice now in the two weeks I've been running this script on my Raspberry Pi. Early in the morning, the Perl script seems to exit without any reason given as to why. I'm running the script as a background process on boot so I don't have the full output from it prior to the crash (although I have now told it to log to a file so next time I should get more information?). I do have the contents of /var/log/eversolar, which I've pasted below:

    2015-09-02 05:09:33: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:09:35: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:09:37: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:09:39: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:09:41: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:09:43: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:09:46: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:09:48: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:09:51: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:09:53: Severity 1, Found serial number: EL36806011540195 16
    2015-09-02 05:09:53: Severity 1, Equal
    2015-09-02 05:09:53: Severity 1, Config Id: 40124
    2015-09-02 05:09:55: Severity 2, No response to 'send register address' request for inverter EL36806011540195
    2015-09-02 05:09:57: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:10:00: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:10:02: Severity 1, Found serial number: EL36806011540195 16
    2015-09-02 05:10:02: Severity 1, Equal
    2015-09-02 05:10:02: Severity 1, Config Id: 40124
    2015-09-02 05:10:04: Severity 2, No response to 'send register address' request for inverter EL36806011540195
    2015-09-02 05:10:06: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:10:08: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:10:10: Severity 1, Found serial number: EL36806011540195 16
    2015-09-02 05:10:10: Severity 1, Equal
    2015-09-02 05:10:10: Severity 1, Config Id: 40124
    2015-09-02 05:10:12: Severity 2, No response to 'send register address' request for inverter EL36806011540195
    2015-09-02 05:10:14: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:10:17: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:10:19: Severity 1, Found serial number: EL36806011540195 16
    2015-09-02 05:10:19: Severity 1, Equal
    2015-09-02 05:10:19: Severity 1, Config Id: 40124
    2015-09-02 05:10:21: Severity 2, No response to 'send register address' request for inverter EL36806011540195
    2015-09-02 05:10:23: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:10:26: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:10:28: Severity 1, Found serial number: EL36806011540195 16
    2015-09-02 05:10:28: Severity 1, Equal
    2015-09-02 05:10:28: Severity 1, Config Id: 40124
    2015-09-02 05:10:30: Severity 2, No response to 'send register address' request for inverter EL36806011540195
    2015-09-02 05:10:32: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:10:34: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:10:36: Severity 1, Found serial number: EL36806011540195 16
    2015-09-02 05:10:36: Severity 1, Equal
    2015-09-02 05:10:36: Severity 1, Config Id: 40124
    2015-09-02 05:10:38: Severity 1, Inverter acknowledged registration
    2015-09-02 05:10:40: Severity 2, No response to 'query inverter id' request for inverter EL36806011540195
    2015-09-02 05:10:42: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:10:44: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:10:46: Severity 1, Found serial number: EL36806011540195 16
    2015-09-02 05:10:46: Severity 1, Equal
    2015-09-02 05:10:46: Severity 1, Config Id: 40124
    2015-09-02 05:10:49: Severity 2, No response to 'send register address' request for inverter EL36806011540195
    2015-09-02 05:10:51: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:10:53: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:10:55: Severity 1, Found serial number: EL36806011540195 16
    2015-09-02 05:10:55: Severity 1, Equal
    2015-09-02 05:10:55: Severity 1, Config Id: 40124
    2015-09-02 05:10:58: Severity 2, No response to 'send register address' request for inverter EL36806011540195
    2015-09-02 05:11:00: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:11:03: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:11:05: Severity 1, Found serial number: EL36806011540195 16
    2015-09-02 05:11:05: Severity 1, Equal
    2015-09-02 05:11:05: Severity 1, Config Id: 40124
    2015-09-02 05:11:07: Severity 1, Inverter acknowledged registration
    2015-09-02 05:11:09: Severity 2, No response to 'query inverter id' request for inverter EL36806011540195
    2015-09-02 05:11:11: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:11:13: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:11:15: Severity 1, Found serial number: EL36806011540195 16
    2015-09-02 05:11:15: Severity 1, Equal
    2015-09-02 05:11:15: Severity 1, Config Id: 40124
    2015-09-02 05:11:18: Severity 2, No response to 'send register address' request for inverter EL36806011540195
    2015-09-02 05:11:20: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:11:22: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:11:24: Severity 1, Found serial number: EL36806011540195 16
    2015-09-02 05:11:24: Severity 1, Equal
    2015-09-02 05:11:24: Severity 1, Config Id: 40124
    2015-09-02 05:11:26: Severity 1, Inverter acknowledged registration
    2015-09-02 05:11:28: Severity 2, No response to 'query inverter id' request for inverter EL36806011540195
    2015-09-02 05:11:30: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:11:32: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:11:34: Severity 1, Found serial number: EL36806011540195 16
    2015-09-02 05:11:34: Severity 1, Equal
    2015-09-02 05:11:34: Severity 1, Config Id: 40124
    2015-09-02 05:11:36: Severity 1, Inverter acknowledged registration
    2015-09-02 05:11:38: Severity 2, No response to 'query inverter id' request for inverter EL36806011540195
    2015-09-02 05:11:40: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:11:42: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:11:44: Severity 1, Found serial number: EL36806011540195 16
    2015-09-02 05:11:44: Severity 1, Equal
    2015-09-02 05:11:44: Severity 1, Config Id: 40124
    2015-09-02 05:11:46: Severity 1, Inverter acknowledged registration
    2015-09-02 05:11:49: Severity 2, No response to 'query inverter id' request for inverter EL36806011540195
    2015-09-02 05:11:51: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:11:53: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:11:55: Severity 1, Found serial number: EL36806011540195 16
    2015-09-02 05:11:55: Severity 1, Equal
    2015-09-02 05:11:55: Severity 1, Config Id: 40124
    2015-09-02 05:11:57: Severity 1, Inverter acknowledged registration
    2015-09-02 05:11:59: Severity 1, Connected to inverter: 1  5000E1.00      TL3680    EL368060115401953600V610-01021-04
    2015-09-02 05:11:59: Severity 2, Asking for any inverters to register
    2015-09-02 05:12:01: Severity 3, No response to 'offline query' request - no offline inverters
    2015-09-02 05:12:03: Severity 2, EL36806011540195 lost contact with inverter (1 time(s))
    2015-09-02 05:12:13: Severity 2, Asking for any inverters to register

solmoller, 2015-09-02:
Hello again Stephen

Could this be related to the unpacking we just fixed in issue #9 ? I see unpack A* a few places on the serial number. 
Unfortunately there is not much time to test as I work long hours presently

I dread you need to stop the background service and run the perl script from a terminal window so we can log perl excemptions. - Or find some way to extract it from your process.

Henrik

solmoller, 2015-09-02:
Oh, and the connect/disconnect is very typical for mornings where we have dim lights making the inverters start and stop a few times.

StephenE, 2015-09-12:
10 days later and it's finally happened again!

    2015-09-12 05:40:46: Severity 1, Inverter acknowledged registration
    2015-09-12 05:40:55: Severity 1, Found serial number: EL36806011540195 16
    2015-09-12 05:40:55: Severity 1, Equal
    2015-09-12 05:40:55: Severity 1, Config Id: 40124
    2015-09-12 05:40:57: Severity 1, Inverter acknowledged registration
    2015-09-12 05:41:06: Severity 1, Found serial number: EL36806011540195 16
    2015-09-12 05:41:06: Severity 1, Equal
    2015-09-12 05:41:06: Severity 1, Config Id: 40124
    2015-09-12 05:41:08: Severity 1, Inverter acknowledged registration
    2015-09-12 05:41:10: Severity 1, Connected to inverter: 1  5000E1.00      TL3680    EL368060115401953600V610-01021-04

This is the output from the perl script:
    35 33 36 30 30 56 36 31 30 2d 30 31 30 32 31 2d 30
    34 0d ba 00
    sending packet to inverter...
    aa 55 01 00 00 00 10 00 00 01 10
    sending packet to inverter...
    aa 55 01 00 00 00 10 00 00 01 10
    received packet from inverter:
    ff
    sending packet to inverter...
    aa 55 01 00 00 00 10 00 00 01 10
    received packet from inverter:
    aa 55 00 00 01 00 10 80 12 45 4c 33 36 38 30 36 30
    31 31 35 34 30 31 39 35 00 01 05 05
    2015-09-12 05:40:44: Severity 1, Found serial number: EL36806011540195 16
    2015-09-12 05:40:44: Severity 1, Equal
    2015-09-12 05:40:44: Severity 1, Config Id: 40124
    sending packet to inverter...
    aa 55 01 00 00 00 10 01 13 45 4c 33 36 38 30 36 30
    31 31 35 34 30 31 39 35 00 01 24 04 ab
    received packet from inverter:
    aa 55 00 24 01 00 10 81 01 06 01 bc
    2015-09-12 05:40:46: Severity 1, Inverter acknowledged registration
    sending packet to inverter...
    aa 55 01 00 00 24 11 03 00 01 38
    received packet from inverter:
    aa 55 00 24 01 00 11 83 4d 31 20 20 35 30 30 30 45
    31 2e 30 30 20 20 20 20 20 20 54 4c 33 36 38 30 20
    20 20 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    00 00 45 4c 33 36 38 30 36 30 31 31 35 34 30 31 39
    35 33 36 30 30 56 36 31 30 2d 30 31 30 32 31 2d 30
    34 0d ba 00
    sending packet to inverter...
    aa 55 01 00 00 00 10 00 00 01 10
    sending packet to inverter...
    aa 55 01 00 00 00 10 00 00 01 10
    received packet from inverter:
    ff
    sending packet to inverter...
    aa 55 01 00 00 00 10 00 00 01 10
    received packet from inverter:
    aa 55 00 00 01 00 10 80 12 45 4c 33 36 38 30 36 30
    31 31 35 34 30 31 39 35 00 01 05 05
    2015-09-12 05:40:55: Severity 1, Found serial number: EL36806011540195 16
    2015-09-12 05:40:55: Severity 1, Equal
    2015-09-12 05:40:55: Severity 1, Config Id: 40124
    sending packet to inverter...
    aa 55 01 00 00 00 10 01 13 45 4c 33 36 38 30 36 30
    31 31 35 34 30 31 39 35 00 01 24 04 ab
    received packet from inverter:
    aa 55 00 24 01 00 10 81 01 06 01 bc
    2015-09-12 05:40:57: Severity 1, Inverter acknowledged registration
    sending packet to inverter...
    aa 55 01 00 00 24 11 03 00 01 38
    received packet from inverter:
    aa 55 00 24 01 00 11 83 4d 31 20 20 35 30 30 30 45
    31 2e 30 30 20 20 20 20 20 20 54 4c 33 36 38 30 20
    20 20 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    00 00 45 4c 33 36 38 30 36 30 31 31 35 34 30 31 39
    35 33 36 30 30 56 36 31 30 2d 30 31 30 32 31 2d 30
    34 0d ba 00
    sending packet to inverter...
    aa 55 01 00 00 00 10 00 00 01 10
    sending packet to inverter...
    aa 55 01 00 00 00 10 00 00 01 10
    received packet from inverter:
    ff
    sending packet to inverter...
    aa 55 01 00 00 00 10 00 00 01 10
    received packet from inverter:
    aa 55 00 00 01 00 10 80 12 45 4c 33 36 38 30 36 30
    31 31 35 34 30 31 39 35 00 01 05 05
    2015-09-12 05:41:06: Severity 1, Found serial number: EL36806011540195 16
    2015-09-12 05:41:06: Severity 1, Equal
    2015-09-12 05:41:06: Severity 1, Config Id: 40124
    sending packet to inverter...
    aa 55 01 00 00 00 10 01 13 45 4c 33 36 38 30 36 30
    31 31 35 34 30 31 39 35 00 01 24 04 ab
    received packet from inverter:
    aa 55 00 24 01 00 10 81 01 06 01 bc
    2015-09-12 05:41:08: Severity 1, Inverter acknowledged registration
    sending packet to inverter...
    aa 55 01 00 00 24 11 03 00 01 38
    received packet from inverter:
    aa 55 00 24 01 00 11 83 4d 31 20 20 35 30 30 30 45
    31 2e 30 30 20 20 20 20 20 20 54 4c 33 36 38 30 20
    20 20 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00
    00 00 45 4c 33 36 38 30 36 30 31 31 35 34 30 31 39
    35 33 36 30 30 56 36 31 30 2d 30 31 30 32 31 2d 30
    34 0d ba
    2015-09-12 05:41:10: Severity 1, Connected to inverter: 1  5000E1.00      TL3680    EL368060115401953600V610-01021-04
    sending packet to inverter...
    aa 55 01 00 00 00 10 00 00 01 10
    received packet from inverter:
    00
    sending packet to inverter...
    aa 55 01 00 00 24 11 02 00 01 37
    sending packet to inverter...
    aa 55 01 00 00 24 11 02 00 01 37
    received packet from inverter:
    ff 00 ff

Hopefully that means something to you. I was hoping to see an exception message or a callstack, but I guess that would be asking too much... :)

StephenE, 2015-09-12:
Whoops, wrong button!

solmoller, 2015-09-12:
FF and 00 could imply sort circuits of some kind. ..

solmoller, 2015-11-03:
Closing due to lack of activity

closed 2015-11-03 by solmoller