solmoller/eversolar-monitor #13: No inverters connected

SolarNL, 2016-02-07:
First of all thank you for the code and clear manual.

Unfortunately I cannot get my setup working, the Eversolar monitor kees saying 'No inverters connected'.

I have a Zeversolar Evershine TL3680, connected with a custom RJ45 to a RS485 to USB connector, see pictures;

[image]
[image]
[image]

The USB is connected to a raspberry pi 2B on which I installed your image file with the monitor. Unfornunately I cannot get it working, tried several different setups of the ini file and experimented with the wiring, but it doesn't seem to work...
Do you maybe have some advice?

wotid, 2016-02-07:
I used the same adapters as you (but then changed to an RJ45 - DB9 adapter so I could use a standard network cable for connecting).

Mine is:

Eversolar 1 TxA to Adapter Rx+ (3)
Eversolar 2 TxB to Adapter Rx- (4)
Eversolar 3 RxA to Adapter Tx+ (2)
Everolar 6 RxB to Adapter Tx- (1)

Adapter 5 & 6 voltage terminals not connected to anything. My Eversolar numbers are for my older TL models and correspond to RJ45 pin numbers.

SolarNL, 2016-02-07:
Thank you for your answer.

According to my inverter manual, it should be wired differently:

[image]

I tried your wiring scheme and I disconnected the voltage wires, but still the same result:

[image]

Actually the RJ45 - DB9 adapter looks like a very nice solution, I will probably also buy one of these once I get it running.

wotid, 2016-02-07:
That's the same Eversolar RJ45 pin out as my inverter (now a TL4600AS, was 2 X TL1500AS)

i,e I shortened TX_RS485A to TxA So if you copy my wiring, you should be good to go.

In the ini file does the [serial] port = your adapter? 

Working old vs working new.
[image]

SolarNL, 2016-02-07:
My wiring is now the same as yours, so that should be ok.

What do you mean by [serial] port = your adapter? I have it on ttyUSB0

[image]

wotid, 2016-02-07:
I'm not 100% sure how you're supposed to do this. Check /var/log/dmesg to make sure your adapter is ttyUSB0 It looks like from your photo you have only the one device plugged into a USB port so it should be ttyUSB0 check anyway for something like: (I have 2 X pl2303 converters but have a udev rule so I know what is what).

[    4.016409] usbserial: USB Serial support registered for pl2303
[    4.016580] pl2303 1-1.3:1.0: pl2303 converter detected
[    4.022510] usb 1-1.3: pl2303 converter now attached to ttyUSB0
[    4.022610] pl2303 1-1.5:1.0: pl2303 converter detected
[    4.027371] usb 1-1.5: pl2303 converter now attached to ttyUSB1

wotid, 2016-02-07:
There may be an issue with your [pvoutput] section too, not sure if it will affect anything, but enabled = 1 and the same inverter is entered twice.

SolarNL, 2016-02-07:
The boot seems to be giong OK:

[image]

I have changed the pv-log and pvoutput to 0

Still no inverter found... :(

wotid, 2016-02-07:
Sorry, I can't offer any more advice. Mine just 'works'. I don't know where you're from, but I'm sure you're doing all your testing while the inverter is alive during the day. Nothing will happen while it is shut down ;)

[image]

SolarNL, 2016-02-07:
Thank you for your help anyway. I'm from the Netherlands and it's getting dark outside now. 

I think I'll buy rj45 - db9 adapter like you and try with another utp cable, because I'm also not convinced of the quality of my cheap Chinese utp cable.

wotid, 2016-02-07:
Good luck, I'm looking forward to a screen shot when it is working. While I'm happy with my 06:38 start up and 20:55 shut down at the moment, during winter it is a lot shorter and sadder :( 

Here's my PVOutput data from eversolar-monitor http://pvoutput.org/list.jsp?sid=12337

I can't even remember how I did this, but I use a MySQL database to store the data every ~30 seconds (my Pi is busy) and retrieve it here http://wotid.dyndns.org/solarhtml/index.php

SolarNL, 2016-02-08:
Yes me too, I'll have to wait some weeks for my adapter, and then I will start trying again. Here it is winter now, summer is a lot better :)

Your set up looks really nice, I see in the picture you have a logger from Zeversolar, where is your pi?

wotid, 2016-02-08:
My Pi is in a bread box inside. I no longer have the 2 X TL1500 inverters, now it's a single TL4600. I don't use the Eversolar logger anymore either. I wanted to upload the inverter data to PVOutput rather than the Zeversolar cloud and found Eversolar Monitor would do the job.

[image]

solmoller, 2016-02-08:
Try to raise debug to 3. ..

Henrik

SolarNL, 2016-02-08:
Thank you for your advise. I will try it in the weekend, because these day's I won't be home when there is sunshine.

SolarNL, 2016-02-09:
I quickly tried it, but without result, I get the following output:

[image]

2016-02-09 08:47:18: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:47:16: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:47:14: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:47:12: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:47:10: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:47:08: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:47:06: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:47:04: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:47:02: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:47:00: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:46:58: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:46:56: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:46:54: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:46:52: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:46:50: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:46:48: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:46:46: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:46:44: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:46:42: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:46:40: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:46:38: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:46:36: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:46:34: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:46:32: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:46:30: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:46:28: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:46:26: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:46:24: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:46:22: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:46:20: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:46:18: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:46:16: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:46:14: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:46:12: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:46:10: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:45:35: Severity 2, Asking all inverters to re-register
2016-02-09 08:45:35: Severity 2, Connecting to the serial port

2016-02-09 08:45:35: Severity 3, Configfile is: eversolar.ini
2016-02-09 08:45:26: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:45:24: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:45:22: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:45:20: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:45:18: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:45:16: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:45:14: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:45:12: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:45:10: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:45:08: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:45:06: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:45:04: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:45:02: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:45:00: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:44:58: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:44:56: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:44:54: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:44:52: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:44:50: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:44:48: Severity 3, No response to 'offline query' request - no offline inverters
[... cut ...]
2016-02-09 08:43:22: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:43:20: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:43:18: Severity 3, No response to 'offline query' request - no offline inverters
2016-02-09 08:42:43: Severity 2, Asking all inverters to re-register
2016-02-09 08:42:43: Severity 2, Connecting to the serial port

2016-02-09 08:42:43: Severity 3, Configfile is: eversolar.ini
2016-01-23 17:08:32: Severity 1, Connecting to the ethernet to serial converter

Reneke43, 2016-05-03:
Hello SolarNL, I don't know if you're still looking for an answer. I've got a TL5000 and working with Eversolar monitor. On a Danish forum I found the connection method for the TL-series.
Make sure you **don't** connect another device such as the Zeversolar PMU.
Use a standard network cable and cut off 1 connector.
Then connect to the RS485 connector as follows:

Orange-white & Green-white together > Tx+
Orange & Green > Tx-
Blue & Blue-white > Gnd

Plug-in the RJ45 connector at your inverter and start the script. After a few seconds (an requests from the script) you'll see the reply. And the result in your browser.

SolarNL, 2016-05-04:
Hi Reneke,

Thank you for your comment, I did not manage to get it working yet so your advice is very welcome.

I will try your wiring scheme this weekend, I see that the RX is not connected?

Reneke43, 2016-05-04:
Yes, that's correct. 6 wires from the inverter are connected as 3 pairs to your RS485 connecter. RX +/- and 5-12V are not used.

SolarNL, 2016-05-04:
Perfect thanks, I will try it directly tonight and let you know!

SolarNL, 2016-05-05:
[image]

The first try; successful! Thank you so much!

solmoller, 2016-05-05:
Good to hear. Thanks for reporting back, and thanks for helping with the
support

Henrik

solmoller, 2016-05-05:
Reneke,

Do you want to be added as admin here?

Henrik
Den 05/05/2016 12.44 PM skrev "Mette og Henrik Jørgensen" <
mettehenrik@lillefamilie.dk>:

Reneke43, 2016-05-05:
@SolarNL Glad I could help.
@solmoller Thanx for your invitation, I have to say though, I'm no programmer but quite god al Googling ;-)

wotid, 2016-05-05:
Wow, you got it working. Congratulations. :)

SolarNL, 2016-05-11:
Now logging live on pvoutput: http://pvoutput.org/intraday.jsp?id=39827&sid=36385&dt=20160509 :)

The temperature it is logging, is it the inverter temperature?

wotid, 2016-05-11:
Yes, it's the inverter temperature.

SolarNL, 2016-05-11:
ok perfect, thanks

closed 2016-05-11 by SolarNL