solmoller/eversolar-monitor #14: Upload to Domoticz

Reneke43, 2016-05-05:
There are several upload options in the script. However, I would like to upload the data to Domoticz. For this you can use JSON to upload the data to virtual hardware.
I'm struggling how to get this done by modifying the Perl script.

solmoller, 2016-12-29:
Sounds like a feasible route you have selected - did you get it to work?

Reneke43, 2017-01-16:
I've been building other automation in Domoticz for my home. (Like a nice theater mode dimming the lights on movie start for example ;-))
Your comment triggered me to continue with this project.
Domoticz has several options to upload data. The script writes JSON text to /tmp/eversolar. This could also be used to modify a dummy electricity sensor by uploading the data to Domoticz with an URL like
http://domoticz_ip:domoticz_port/json.htm?type=command&param=udevice&idx=IDX&nvalue=0&svalue=POWER;ENERGY

solmoller, 2017-01-16:
> param=udevice&idx=IDX&nvalue=0&svalue=POWER;ENERGY

If you get it to work, please submit the code

BR

Henrik

Reneke43, 2017-01-20:
I got it working. I added a section Domoticz to the .ini file and the script. Also added a small instruction to the README.md. Also I edited the index.html so that it updates every 10 seconds. A bit more realtime ;-)
I've added my working eversolar folder, so there is data included from my inverter. I don't know how to wipe it, but I guess you do.
For now it only uploads realtime generated power. I will continue to work to add more data sensors to Domoticz. (Current, Voltage, Inverter temperature...)
[eversolar-monitor-code.zip](https://github.com/solmoller/eversolar-monitor/files/720300/eversolar-monitor-code.zip)

solmoller, 2017-01-21:
Thanks for submitting the code, will you be progressing shortly? I'll definately add it to the code, it's good to have the project back. Incidentally, github offers quite substantial support for having multiple developers sharing a project, you might want to look into how it works.

ronneman, 2017-05-28:
@Reneke43 ,

Any progress on the Domoticz integration. I am looking forward to this functionality as well.

solmoller, 2017-10-31:
So this should be added to the ini:
`# upload to Domoticz home automation server (more information: https://www.domoticz.com)
enabled = 0
address = "x.x.x.x"                     
# Domoticz server ip. Can be localhost when running this script on the same system as this script.
port = 8080                             
# Domoticz port (8080 is default)
IDX = xxx                               
# virtual sensor IDX number. This can be found on the Domoticz hardware page.

`

And this to the pl:
`
$config->define("domoticz_enabled=s");
$config->define("domoticz_address=s");
$config->define("domoticz_port=s");
$config->define("domoticz_IDX=s")

 ##  Data to Domoticz
            ##
            ##
            ##
            ###############################################################################

            if($config->domoticz_enabled) {

                my $domoticz_address = $config->domoticz_address;
                my $domoticz_port = $config->domoticz_port;
                my $domoticz_IDX = $config->domoticz_IDX;

                my $cmd = `curl -s "http://$domoticz_address:$domoticz_port/json.htm?type=command&param=udevice&idx=$domoticz_IDX&nvalue=0&svalue=$pac;$e_today_wh" `;
		chomp($cmd);
                pmu_log("Severity 3, ".$inverters{$inverter}{"serial"}." uploading to domoticz, response: $cmd");

            }

            
`

Is that all, or did you change other places?

solmoller, 2017-12-03:
Updated the code, can you test it, please?

Reneke43, 2017-12-03:
I've just downloaded the code. I'll test tomorrow when I have some daylight over here :-)

solmoller, 2017-12-03:
Great. Should I put your name in the code?

solmoller, 2017-12-28:
I have added your name to the code as "Reneke43", If you'd like your full name in let me know.

Ctenberge, 2018-02-01:
I tried to add Domoticz and succeeded mostly, no errors in the eversolar section, only I get never data in the dummy device. 
Domoticz runs on another Pi and I am a bit confused here. I followed your instructions for the eversolar.ini file and do not understand enabled should be 0 (zero) where normally one would expect a 1 (one).
I tried both by the way and did'nt get any result. No power is being shown.
What is missing?

Ctenberge, 2018-02-01:
I found it. For some reason this line was commented
 my $cmd = `curl -s "http://$domoticz_address:$domoticz_port/json.htm?type=command&param=udevice&idx=$domoticz_IDX&nvalue=0&svalue=$pac;$e_today_wh" `;

now it works.
Thanks

solmoller, 2018-02-03:
Thanks for the fix, version 0.171 released with your name in it

closed 2018-02-03 by solmoller

Ctenberge, 2018-02-04:
Hello Henrik,

the quotes *in*  the expression are neccesarily and must therefore be
present.
I did'nt get Domoticz at work first and tried several ways to get the job
done, and I commented the whole line.
Later on I forget about it, removed my efforts and it still did'nt work.
So I posted my problem and shortley afterwards discovered the remaining
qoute at the beginning of the line (a #).
After removing, Domoticz  started the data from eversolar.
I posted the result but failed to tell about the # , sorry.

with regards, Carel ten Berge

2018-02-03 11:59 GMT+01:00 Henrik Møller Jørgensen <notifications@github.com