solmoller/eversolar-monitor #16: Zeversolar to RS485 to UART adapter

fieldhof, 2016-08-01:
I've bought a RS485 to UART adapter from ebay [http://www.ebay.com/itm/361375498598?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT](url)

On the RS485 side of the board there are three connections, A+, B- and GND

But i'm curious how to hook this thing up to the Zeversolar inverter

From the manual I got this table

Ethernet pin | Function
 --- | --- |
1 | TX_RS485A|
2 |TX_RS485B|
3|RX_RS485A|
4|GND|
5|GND|
6|RX_RS485B|
7|+7V|
8|+7V|

From the guide of this github i got this table

|Eversolar pin|Eversolar signal|RS485|
|:------------|:---------------|:----|
|1 |RXD+ |Data + |
|2 |RXD- |Data – |
|3 |TXD+ |Data +|
|4 |gnd | GND |
|6 |TXD- |Data – |

How do i connect the inverter to the board?

Ethernet pin | Function | RS485
 --- | --- | ---
1 | TX_RS485A | A+/B-
2 |TX_RS485B | A+/B-
3 |RX_RS485A | A+/B-
4 |GND | GND
5 |GND | GND
6 |RX_RS485B | A+/B-
7 |+7V | NC
8 |+7V | NC

Reneke43, 2016-08-01:
Hello fieldhof,

Try my solution that I posted in the thread "no inverter connected"

"Hello SolarNL, I don't know if you're still looking for an answer. I've got a TL5000 and working with Eversolar monitor. On a Danish forum I found the connection method for the TL-series.
Make sure you don't connect another device such as the Zeversolar PMU.
Use a standard network cable and cut off 1 connector.
Then connect to the RS485 connector as follows:

Orange-white & Green-white together > Tx+
Orange & Green > Tx-
Blue & Blue-white > Gnd

Plug-in the RJ45 connector at your inverter and start the script. After a few seconds (an requests from the script) you'll see the reply. And the result in your browser."

Reneke43, 2016-08-01:
The other connections are not used.

fieldhof, 2016-08-01:
Hi @Reneke43,

Thanks for your help, there are no Tx+ and Tx- on my RS485 board, but I figured it out.
Orange white + Green white > B-
Orange + Green > A+

I currently have 2 TL3000 at my home on different spots and trying to connect 2 esp8266 with both a  RS485 board to the TL3000.

Right now I have a Arduino working with the RS485 board and the TL3000 so soon I'll be going to work on the esp8266. I'm trying to get the esp8266 to work because they are much cheaper than a raspberry pi.

fieldhof, 2016-08-01:
Out of curiosity, how much current can the +7V deliver? And I measured +10V at the wires, but that's no problem in my case

closed 2016-08-04 by fieldhof

Ravindrakant, 2016-12-21:
Hello Sir ,I want to know if I connect the Zerversolar inverter to my Laptop through a rs485 to rs232 converter what kind of data I can get from on the serial termial.Will it be in the form displayed on the LCD or something different .I want to write a code in arduino so that I can monitor the data on the serial port.
Thank you .please help me

solmoller, 2016-12-29:
Your request  may be quite some way outside this project. You can learn the
protocol from the source code, maybe the original base of the project could
be another starting point here:
http://www.sjcnet.id.au/computers/eversolar-inverter-monitoring-with-linux/comment-page-1

Best regards

Henrik

phungorquan, 2019-08-26:
Hello everybody, i was stuck with 3000tl zeversolar, when i connected the usb-rs485 to rj45 of inverter, i didn't receive anything. I tried to send some command afterthat but i also didn't receive.

About the wire, i tried both of way below but nothing happend:

Orange-white & Green-white together > Tx+ (A)
Orange & Green > Tx- (B)

Orange white + Green white > B-
Orange + Green > A+

Could anybody helps me :(

solmoller, 2019-08-26:
What adapter are you using?

Br Henrik

phungorquan, 2019-08-26:
I by this in my country (VietNam) : https://thegioiic.com/products/mach-chuyen-doi-usb-to-rs485

I don't know why the inverter doesn't response anything although i tried to send many commands continously. I used hercules terminal to read serial from the usb.

I tried to commute 2 usb-rs485 each other, it can normally send and receive.

The inverter is on grid and operate normally

phungorquan, 2019-08-26:
This is the family zeversolar inverter document: https://drive.google.com/file/d/1kzXyG2114c4oZBB9dWruK63_X4J58sIg/view?usp=drivesdk

I followed step by step , re-register by send the command x3 , then send query ,... but i can not receive anything

solmoller, 2019-08-26:
That's a difficult one. You have debug set so that you see the sending and
receiving?
Are there any LED's flashing in any side of the communication?

BR Henrik

solmoller, 2019-08-26:
I have moved your comments to a new issue - it's difficult to support on closed issues

phungorquan, 2019-08-27:
Thanks too much, i think it does's have any led on the inverter side, on the usb-rs485 i saw the tx led was blink.
When i plug the rs485 into inverter for the first time, it sent some noise signals to my computer, that's mean my wires aren't broken

solmoller, 2019-08-29:
Are you using your own software of the perl software in this project?

solmoller, 2019-08-29:
I see that you have closed the issue