solmoller/eversolar-monitor #18: Not stable

GPellemans, 2017-02-23:
Hello, I'm using Eversolar-monitor since (I think) January 2014. I'm very happy with it. Thanks.

But from the beginning it isn't very stable. The reading of the eversolar works fine. I can see that every time there is a new reading in tmp/eversolar. 
But to show the reading in the webpage does not always work; not on my intern IP, nor on my extern IP-adres; sometimes it works fine, another moment it is "hanging". 
My workaround is to reboot the raspberry or to kill perl en restart eversolar.pl (with ssh).

Is there anyone with the same experience?
Is there a solution for my problem?

I have tried several things, like stop showing the logmessages, stop sending data to PV-log, do a reading once a minute, debug 0, etc., but nothing helps.

The latest I tried is to follow the instructions to change eversolar.pl with Prefork. On my raspberry I gave the order: 
sudo cpan HTTP::Server::Simple::CGI::PreFork 
I don't know whether that was OK, but the raspberry gave no error.

Result: the program doesn't know tmp/eversolar any more (cat /tmp/eversolar). Also I have no connection using my IP. So I changed it back to the former settings, and accept that it isn't working always.
And I don't know whether "use File::Basename;" is correct. It isn't a line in the normal eversolar.pl. 

See under: 

Has anyone an idea how to fix my problem?
------------------------------------------------------------------------------
Part of eversolar.pl
The ## are the new lines in eversolar (but not working), the other are the old lines (working, but not stable). The lines are under "Web server package" in eversolar.pl.
Just for the reading there is an ' before ##.

 if($config->web_server_enabled)
 {

'##  @@ -176,8 +177,8 @@ if($config->web_server_enabled) {
'##  {
'##
    package SolarWebServer;

'##    use HTTP::Server::Simple::CGI::PreFork;
'##    use base qw(HTTP::Server::Simple::CGI::PreFork);
    use HTTP::Server::Simple::CGI;
    use base qw(HTTP::Server::Simple::CGI);
    use JSON;
    use AppConfig;
'##    use File::Basename;

solmoller, 2017-02-23:
Are you using the SD card image?
Is your raspberry updated and upgraded?

I'm not much into the external IPV6 stuff, but a use HTTP::Server::Simple::CGI; appears to be missing

I've had that image run for more than half a year without problems.. but I hardly ever use the internal webpage, so unless you have a corrupt download, this difference may explain the behaviour.

Unfortunately I cannot test much presently, as the sun is down when I'm back from work. But I'd recommend a reboot, and then reloading the webpage heavily... use 'top' command on the raspberry to monitor processes.

If you reload the image, remember to take a backup of the db.sqlite file. and patch it back in, else you'll lose your history. (hopefully you're already signed up with pvoutput)

BR and good luck

Henrik

Reneke43, 2017-02-23:
In the beginneling I also had stabiliteit issues. What Raspberry Pi do you use? For a Pi 2 I recommend at least a 2 amp power adapter. If you use a Pi 3 you are required to use at least a 3 amp adapter.

wotid, 2017-02-23:
I can't remember my exact symptoms now because it was so long ago, but I have cron kill perl at 5AM every morning and restart eversolar-monitor. I have no problems at all, but somewhere along the way (after having cron kill perl) I disabled the eversolar-monitor webserver and use highcharts to read the database and display the data.
[image]

Regards,

Steve.

solmoller, 2017-02-24:
Hi Steve... Highcharts? nice graphs! I have just added 365 days rolling averages to the code, but I only display it as text. If I'm going to redo any graphics, then highcharts looks pretty good. (365 days rolling graph should give an idea about panel degradation)

Can your graphs be implemented in the project?

Henrik

GPellemans, 2017-02-27:
Thank you for all your reactions.

@Henrik: No I didn't use the SD-card image. After your reaction I took the SD-card image en put the SD-card in my raspberry. It worked, but also that wasn't stable. 
And yes, raspberry is updated and upgraded.
@Reneke43: I use an raspberry from 2013, model 1B. I use a 2 amp adapter. I understand that model Pi 3 at this moment is the newest model. Would it help to switch to that model?
@Steve : every night at 3.30 AM the raspberry gets a reboot. 
I must say that the highcharts-graph looks very impressive. I wonder if there is a way that I can use those highcharts too. 

Since a few weeks I use the app RaspiCheck. With that app I can very easy reboot my raspberry when the  webserver "hangs".

GPellemans, 2017-02-27:
Sorry, wrong button. It wasn't my intention to close the issue yet.

solmoller, 2017-02-28:
Strange.  Any hints from logfiles?  I use 1B as well, and we have many
users without problems. Try swapping out Rpi or power supply

Henrik

closed 2017-07-09 by solmoller