solmoller/eversolar-monitor #19: Power right now

ronneman, 2017-05-29:
Hi all,

Since a few weeks I have an issue that the "Power right now" and "Max power today" are always 0. All the other attributes AC / PV & inverter details are present and updated. However I do think I am missing one string in the inverter details. I have a TL2000-20 with 2 strings.

My cable is a standard network cable with one end changed according to the specs of the introduction page. A USB serial converter and a Rasperry PI B. As i said it did work for a couple of years with all values but suddenly is missing some data.

Any clues on what it might be? 

Best regards,
Ron

solmoller, 2017-05-29:
No. Pls check, if no other values intermittent fail. If it is only that one
value, then you can rule out cabling and start worrying about your system:
 maybe the system actually have production failures. Is it just one of the
two inverters with this behaviour?

BR Henrik

ronneman, 2017-05-29:
No the values are always 0. I have only one inverter but with 2 input strings from the panels.

[image]

ronneman, 2017-05-29:
I shutoff AC & DC power to the inverter, reconnected the strings, turned back the power on and it is working again. The power and max power are listed again.

Thanks for pointing me to the system. Was assuming it was my bad cable making skills ;)

closed 2017-05-29 by ronneman