solmoller/eversolar-monitor #21: markdown formatting at wiki/Hardware.md is broken; URL updates

jpluimers, 2017-07-07:
At https://github.com/solmoller/eversolar-monitor/blob/wiki/Hardware.md some markdown is broken:

- Tables in https://github.com/solmoller/eversolar-monitor/blob/wiki/Hardware.md#cabling
- Code in https://github.com/solmoller/eversolar-monitor/blob/wiki/Hardware.md#inverted-polarity
- Title in https://github.com/solmoller/eversolar-monitor/blob/wiki/Hardware.md#known-good-usb-to-rs22-ports (I guess the title should be `Known good USB to RS422 ports`)
- Adding links to the products (preferably with web.archive.org archival links) will make it easier for users to find the right USB adapters.

solmoller, 2017-07-09:
Hi Jeroen, 

Thanks! All should be updated now, though I am unable to update the product list, as I have no references to adapters from this day and age. 
(It appears people are able to buy adapters and make the solution work, as we have no support issues, and I see new systems going online)

Henrik

closed 2017-07-09 by solmoller

jpluimers, 2017-07-09:
Thanks. I'm about to try this out, so when I've got it working I will put in a pull-request with updated information.