solmoller/eversolar-monitor #22: Connection to ZeverSolar TLC6000

MynahPL, 2017-07-29:
I have a ZeverSolar TLC6000 module installed. It's connected to my home Wi-Fi with static IP address assigned. The inverter has a tiny built-in web server accessible via any browser. 
How should the script be modified to get it to work? I can't get any data so far.

solmoller, 2017-07-29:
How as in how to edit the configuration file or how as what to put in the
file?
 Did you find the file and are you able to edit it?
BR Henrik

MynahPL, 2017-07-29:
Sorry, how as - what to put in there. Is it possible to get access to this data connecting to IP address of the inverter?

solmoller, 2017-07-29:
Set debug to 3 an tryout strings as 1 or 2

Post the output of the logging here

I haven't heard of the 6000 before that I know of, but the manufacturer was
very helpful back when we got this working, so with a little patience we
might get it to include your adapter as well.

Br
Henrik

MynahPL, 2017-07-29:
I'll try that in the morning when it gets back online. Just to make it clear, I'm trying to reach it without serial connection (maybe it's wrong from the beginning). So the inverter (with static IP pre-assigned) is connected to my router with Wi-Fi connection, what I'm trying to do is to get access to it via RaspberryPi (obviously connected to the same network) using all this info described in here.

wotid, 2017-07-30:
The inverter is probably trying to send data to the ZeverCloud, like the PMU's do. You may have to use Wireshark or something to sniff out what data is being transmitted and then write something that will intercept it, format and store it via the RPi.

If I remember correctly, this software was originally based on Steve Cliffe's script he came up with by intercepting the communication between his inverter and PMU, this would be different than the data sent from the PMU to the cloud - the data you will be trying to intercept.

I wonder if ZeverSolar has published the API for ZeverCloud? It would help.

MynahPL, 2017-07-30:
Here is the API documentation for ZeverCloud:
https://www.dropbox.com/s/kiraxlseqpq280f/Solarcloud-API-guide_En_v20141114.pdf?dl=0
I hope it helps.

My logs are below - the first attempt with strings set to "1", later one is after changing it to "2".

2017-07-30 07:55:29: Severity 3, Configfile is: eversolar.ini

2017-07-30 07:55:29: Severity 3, Checking database
2017-07-30 07:55:29: Severity 1, Connecting to the ethernet to serial converter
2017-07-30 07:55:29: Severity 1, Connected to the ethernet to serial converter
2017-07-30 07:55:39: Severity 2, Asking all inverters to re-register
2017-07-30 08:12:21: Severity 3, Configfile is: eversolar.ini

2017-07-30 08:12:21: Severity 3, Checking database
2017-07-30 08:12:21: Severity 1, Connecting to the ethernet to serial converter
2017-07-30 08:12:21: Severity 1, Connected to the ethernet to serial converter
2017-07-30 08:12:31: Severity 2, Asking all inverters to re-register

MynahPL, 2017-07-30:
Sending such request to inverter 
`wget -qO- http://192.168.1.9/home.cgi?sid=0`
results in response as follows https://pastebin.com/TTfki3v9 ,where line:
1,2 - unknown
3 - Registry ID
4 - Registry Key
5 - Hardware Version
6 - Software version
7 - Time
8 - Communication status with ZeverCloud
9 - unknown
10 - SN.
11 - Pac(W)
12 - E_Today(KWh)
13 - Status
14 - unknown

Information is refreshed every 30s.

solmoller, 2017-07-30:
This looks like good usable data. We'll probably be able to get it to work.

Br Henrik

MynahPL, 2017-08-04:
Perfect solution would be to have it worked regardless of ZeverCloud. It would be great to pull data directly from the inverter and then have it stored in DB.

solmoller, 2017-08-04:
It is definately possible to do this - I don't have time to look into this
presently though - maybe you should try yourselves?

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virusfri.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

closed 2018-08-03 by solmoller