solmoller/eversolar-monitor #23: Upload to InfluxDB

sorentorp, 2017-10-09:
Hi (hej) Solmoller and the rest of the project.

I would like to migrate all my historical data stored ind the sqlite3 database to InfluxDB. I think I'll be able to do that with sqlite-to-csv, maybe with the help of some JSON.

But I would also like to modify ever-solarmonitor to keep adding data to the Influx database, so that I can use Grafana to visualize my production. Maybe even ind combination with this dashboard. https://grafana.com/dashboards/2315

So my question is, how would I go about adding it to the script? Would it be possible for you to add a "custom script" option? My cmd would be something like this:

`curl -is -XPOST "http://xxx.xxx.xxx.xxx:8086/write?db=eversolar" --data-binary "$serial_number_Stats,Type=TimeStamp=$timestamp,PAC=$pac,EToday=$e_today,ETotal=$e_total,VPV=$vpv,VPV2=$vpv2,IPV=$ipv,IPV2=$ipv2,VAC=$vac,IAC=$iac,Frequency=$frequency,Temp=$temp" >/dev/null 2>&1`

Any help is appreciated!

solmoller, 2017-10-09:
Hej Søren,

As for the export to CSV, the wiki has a suggestion:

sudo sqlite3 -csv -header -separator ';' db.sqlite "select serial_number,max(e_today), date(timestamp) from inverter group by serial_number,date(timestamp);" > daily.csv

That should export daily values to your database.

When it comes to InfluxDB, you're done the most of the job. I suggest we add InfluxDB as a new export format. The ini file needs to have the URL added, and then you simply copy and modify this section of the code to include your curl sentence:

            ###############################################################################
            ##
            ##
            ##
            ##	Data to PVoutput.org
            ##
            ##
            ##
            ###############################################################################

            if($config->pvoutput_enabled && ($min % $config->pvoutput_status_interval_mins) == 0 && $min != $last_min) {
                my $pv_date = `date +%Y%m%d`;
                my $pv_time = `date +%H:%M`;
                chomp($pv_date);
                chomp($pv_time);

                my $pvoutput_api_key = $config->pvoutput_api_key;
                my $pvoutput_add_status_url = $config->pvoutput_add_status_url;
                my $this_pvoutput_system_id = $config->pvoutput_system_id->{$inverters{$inverter}{"serial"}};
              #use Data::Dumper;
              #print Dumper($this_pvoutput_system_id);
              #die;

                my $cmd = `curl -m 30 -s -d "d=$pv_date" -d "t=$pv_time" -d "v1=$e_today_wh" -d "v2=$pac" -d "v5=$temp" -d "v6=$volts" -H "X-Pvoutput-Apikey:$pvoutput_api_key" -H "X-Pvoutput-SystemId:$this_pvoutput_system_id" $pvoutput_add_status_url 2>&1 `;                chomp($cmd);
                pmu_log("Severity 3, ".$inverters{$inverter}{"serial"}." uploading to pvoutput.org, response: $cmd");

            }

            
When that is working, maybe we should look at updating the web page in the system to reflect the database and dashboard.

BR

Henrik

sorentorp, 2017-10-10:
I'll get started on it, one of these days :-)

sorentorp, 2017-10-12:
Hi again Henrik

So I've worked on the extra module for eversolar-monitor, but no data seems to be written to the database. So I'm hoping you can help me out.

The debug from console is: note that I dont get any response print from influx. When I run my command with dummyvalues instead of $marks, it works fine. So I think that is where the problem lies.

```
2017-10-12 09:15:53: Severity 3, B983000A128P0120 output: 84 W, Total: 13784.5 kWh, Today: 0.06 kWh, 365 days : 1351.6
2017-10-12 09:15:53: Severity 3, B983000A128P0120 uploading to influx, response:
2017-10-12 09:15:54: Severity 3, B983000A128P0120 uploading to pvoutput.org, response: OK 200: Added Status
```
added to eversolar.ini
```
[influx]                              # upload generation to influxdb
enabled = 1
status_interval_mins = 5                # 5 or 10 minutes, depending on your pvoutput.org setting
# system_id = B983000A128P0120
```

added to eversolar.pl
```
$config->define("influx_enabled=s");
$config->define("influx_status_interval_mins=s");
```
and

```
###############################################################################
        ##
        ##
        ##
        ##      Data to InfluxDB
        ##
        ##
        ##
        ###############################################################################

        if($config->influx_enabled && ($min % $config->influx_status_interval_mins) == 0 && $min != $last_min) {
      #      my $pv_date = `date +%Y%m%d`
      #      my $pv_time = `date +%H:%M`;
      #      chomp($pv_date);
      #      chomp($pv_time);

            my $cmd = `curl -is -XPOST "http://192.168.1.6:8086/write?db=eversolar" --data-binary "inverter_Stats,Type=Energy,Interface=$serial_number PAC=$pac,EToday=$e_today,ETotal=$e_tota$
            pmu_log("Severity 3, ".$inverters{$inverter}{"serial"}." uploading to influx, response: $cmd");

        }
```

Any help is appreciated!

sorentorp, 2017-10-12:
Just to followup, a testrun with dummyvalues, gives me the following response.

```
curl -is -XPOST "http://192.168.1.6:8086/write?db=eversolar" --data-binary "Test_Stats,Type=Energy,Interface=invertertest PAC=2000,EToday=3000,ETotal=$5000,VPV=50,VPV2=230,IPV=20,IPV2=30,VAC=40,IAC=50,Frequency=50,Temp=20"
HTTP/1.1 204 No Content
Content-Type: application/json
Request-Id: 8fc28cf0-af47-11e7-b723-000000000000
X-Influxdb-Version: 1.3.5
Date: Thu, 12 Oct 2017 12:19:14 GMT
```

sorentorp, 2017-10-16:
Okay, process so far. I made this one work, but it only drags out some of the data - I would like the rest

This works fine:
`my $cmd = `curl -is -XPOST "http://192.168.1.6:8086/write?db=eversolar" --data-binary "inverter_Stats,Type=Energy,Interface=inverter PAC=$pac,EToday=$e_today_wh,ETotal=$e_total,Temp=$temp"`;                chomp($cmd);`

This command fails due to missing data.
`my $cmd = `curl -is -XPOST "http://192.168.1.6:8086/write?db=eversolar" --data-binary "inverter_Stats,Type=Energy,Interface=inverter PAC=$pac,EToday=$e_today_wh,ETotal=$e_total,VPV=$vpv,VPV2=$vpv2,IPV=$ipv,IPV2=$ipv2,VAC=$vac,IAC=$iac,Frequency=$frequency,Temp=$temp"  `;                chomp($cmd);`

Error message:
`{"error":"unable to parse 'inverter_Stats,Type=Energy,Interface=inverter PAC=29,EToday=30,ETotal=13798.7,VPV=,VPV2=,IPV=,IPV2=,VAC=,IAC=,Frequency=,Temp=11': missing field value"}`

solmoller, 2017-10-19:
Looks like simple syntax,

Is there a comma missing here?

'my $cmd = curl -is -XPOST "http://192.168.1.6:8086/write?db=eversolar" --data-binary "inverter_Stats,Type=Energy,Interface=inverter COMMA PAC=$pac,EToday=$e_today_wh,ETotal=$e_total,Temp=$temp"; chomp($cmd);'

BR 
Henrik

sorentorp, 2017-10-20:
I got it working, and have made it upload all the available data - as far as I can tell... Where do you want the code?

I'm sure we could make it a bit more "userfriendly", so that coming users can setup the server address and port in the config.

Sofar it can be found here: 
https://pastebin.com/4qXmwJvR

solmoller, 2017-10-21:
Hej Søren 

Great! 

The  ini file is the way to go enabling and managing the different setups people have.

I can't see the fork, but pastebin it or likewise, and I'll have a look. We'll definately add your code to the project when it is ready.

I've just realized that I have  a bug in the code: The daily productiond I use for rolling 365 dayd production is deleted on reboot, My Raspberry is never rebooted, so it did not pop up here. 

Actally, this line;
Last 365 days production | 1902.9kWh
has become the main place for me to look, as it gives me an idea of system degradation over time.

Graphing that should be added to the solution

Anyway, let me know how it proceeds

Henrik

sorentorp, 2017-10-24:
I havent got around to coding, but I have made some work on the dashboard.

I have Grafana presenting all the data I'm pulling - I haven't pulled the "Last 365" yet, I'm get that also. I have also added the weather forecast - which is interestin, esspecially the "percentage of the sky with sun".

Here is a view at the dashboard, as it is right now. (7 days access) 
[ http://xn--srentorp-54a.dk:3000/dashboard/snapshot/9ICfNVjBWfP4n4Ww15YXDIjJdaiVNUEY](url)

Zathras42, 2017-10-24:
Nice dashboard.
What would I need to do to display my TL1500 monitor like this?

sorentorp, 2017-10-24:
If it works with eversolar-monitor, this will also work for you...

sorentorp, 2017-10-25:
@solmoller  I have added my changes to https://github.com/sorentorp/eversolar-monitor

How do I call "Last 365 days production" - I'm asking because you commented that it is faulty at the moment.

solmoller, 2017-10-26:
There's a separate table in the database, daily

Unfortunately I think the code deletes the table at each restart, I need to update the start of the perl script

sorentorp, 2017-10-27:
Okay - let me know when you have it fixed, I'll merge that in

solmoller, 2017-10-31:
I think it is enough to comment out this line
`$dbh->do("drop table daily");`

solmoller, 2017-12-03:
The code was working, but clumsy. I have fixed that, and released a new version today, look for 365 - I'm not going by year, but by 365 days to avoid leap years having more production.

solmoller, 2018-01-21:
Fixed, closing

closed 2018-01-21 by solmoller

sorentorp, 2018-01-21:
Thank you. 

I had some trouble using your new version, with my additions - and havent had the time to look into it. So I'm still on the old one.