solmoller/eversolar-monitor #24: Zeversolar ZL-1500

blickdicht, 2017-10-18:
Hi,

I am trying to get it to work on a Zeversolar ZL-1500. I was convinced, that the protocol will be compatible since Zeversolar seems to be a combination of Eversolar and ZOF:
_"Zeversolar combines Eversolar's low and medium power inverters and ZOF's high power inverters"_
But it seems a little more complicated:

Hardware is a raspi where I installed the monitor successfully manually (great documentation!). Since there is already a zigbee domoticz system running on the same machine, I could not use the image. 

The inverter is connected via 5m cable to a FTDI USB/485 converter on the Raspberry. There is clearly communication going on. The TX and RX LED blink happily: a lot of TX and some of the packets answered by RX. I experimented with line termination on one or both sides to see if it would help, but I keep getting 
`Severity 3, No response to 'offline query' request - no offline inverters`

Since the logfile is not verbose enough I streamed the console output to a text file:

`2017-10-18 17:17:12: Severity 3, Configfile is: eversolar.ini`

`Update old database version, or print 2 fail messages : `
`Done updating old database version.`
`2017-10-18 17:17:12: Severity 3, Checking database`
`2017-10-18 17:17:12: Severity 2, Connecting to the serial port`
`2017-10-18 17:17:13: Severity 2, Asking all inverters to re-register`
`sending packet to inverter... `
`aa 55 03 00 00 00 10 04 00 01 16 `
`sending packet to inverter... `
`aa 55 03 00 00 00 10 04 00 01 16 `
`sending packet to inverter... `
`aa 55 03 00 00 00 10 04 00 01 16 `
`sending packet to inverter... `
`aa 55 03 00 00 00 10 04 00 01 16 `
`sending packet to inverter... `
`aa 55 03 00 00 00 10 04 00 01 16 `
`sending packet to inverter... `
`aa 55 03 00 00 00 10 04 00 01 16 `
`sending packet to inverter... `
`aa 55 03 00 00 00 10 04 00 01 16 `
`sending packet to inverter... `
`aa 55 03 00 00 00 10 04 00 01 16 `
`sending packet to inverter... `
`aa 55 03 00 00 00 10 00 00 01 12 `
`received packet from inverter: `
`aa 55 01 00 00 00 10 00 00 01 10 aa 55 00 00 01 00 `
`10 80 12 42 53 31 35 30 30 36 30 31 31 37 34 30 32 `
`37 30 03 06 05 02 aa 55 01 00 00 00 10 01 11 42 53 `
`31 35 30 30 36 30 31 31 37 34 30 32 37 30 f8 05 71 `
`aa 55 00 f8 01 00 10 81 01 06 02 90 aa 55 01 00 00 `
`f8 11 03 00 02 0c aa 55 00 f8 01 00 11 83 93 31 20 `
`20 32 30 30 30 45 31 2e 30 30 20 20 20 20 5a 4c 20` 
`31 35 30 30 53 20 20 20 20 20 20 20 20 5a 45 56 45 `
`52 53 6f 6c 61 72 20 20 42 53 31 35 30 30 36 30 31 `
`31 37 34 30 32 37 30 33 36 30 30 56 36 31 30 2d 30 `
`31 30 33 37 2d 30 39 00 00 00 00 00 00 00 00 00 00 `
`00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 `
`00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 `
`56 36 31 30 2d 30 31 30 33 38 2d 30 39 56 36 31 30 `
`2d 31 30 30 30 30 2d 30 33 18 94 aa 55 01 00 00 f8 `
`11 `
`2017-10-18 17:17:32: Severity 3, No response to offline query request - no offline inverters`
`sending packet to inverter...` 
`aa 55 03 00 00 00 10 00 00 01 12 `
`received packet from inverter: `
`00 00 02 09 aa 55 00 f8 01 80 aa 55 01 00 00 f8 11 `
`00 00 02 09 aa 55 01 00 00 f8 11 00 00 02 09 aa 55 `
`01 00 00 f8 11 00 00 02 09 aa 55 01 00 00 00 10 00` 
`00 01 10 aa 55 01 00 00 f8 11 02 00 02 0b ff aa 55 `
`01 00 00 f8 11 02 00 02 0b aa 55 01 00 00 f8 11 02 `
`00 02 0b aa 55 00 f8 01 00 11 82 32 01 14 01 1e 04 `
`76 00 00 00 06 00 00 00 05 09 04 13 87 00 40 ff ff `
`00 00 01 59 00 00 00 94 00 01 00 00 00 00 ff ff 00 `
`00 00 00 00 00 00 00 ff 00 00 00 0a 47 aa 55 01 00 `
`00 00 10 00 00 01 10 aa 55 00 00 01 00 10 80 12 42` 
`53 31 35 30 30 36 30 31 31 37 34 30 32 37 30 03 06 `
`05 02 aa 55 01 00 00 00 10 01 11 42 53 31 35 30 30 `
`36 30 31 31 37 34 30 32 37 30 f8 05 71 aa 55 00 f8 `
`01 00 10 81 01 06 02 90 aa 55 01 00 00 f8 11 03 00 `
`02 0c aa 55 00 f8 01 00 11 83 93 31 20 20 32 30 30 `
`30 `
`2017-10-18 17:17:33: Severity 3, No response to 'offline query' request - no offline inverters`
... and so on

It seems to me, they speak a totally different language. If I assume, that packets start with aa 55, then the received packets are not interpreted at all and are simply broken into pieces after 256 Bytes.
Can you make sense of those answers?

Would the inverter stream its data just by itself without initialization or does it react to TX as the LED-blinking on the FTDI suggests? 

I can set the ID of the inverter. Should I, or is it discovered automatically?

Any help is appreciated...

Thanks,
Georg

solmoller, 2017-10-19:
Interesting, here is what I notice:

First an example of well formed communication:

'sending packet to inverter...
aa 55 01 00 00 24 11 03 00 01 38
received packet from inverter:
aa 55 00 24 01 00 11 83 4d 31 20 20 35 30 30 30 45
31 2e 30 30 20 20 20 20 20 20 54 4c 33 36 38 30 20
20 20 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 45 4c 33 36 38 30 36 30 31 31 35 34 30 31 39
35 33 36 30 30 56 36 31 30 2d 30 31 30 32 31 2d 30
34 0d ba 00'

The I have your reply with added line breaks::

`sending packet to inverter...
aa 55 03 00 00 00 10 00 00 01 12
received packet from inverter:
aa 55 01 00 00 00 10 00 00 01 10 
aa 55 00 00 **01 00** 10 80 12 42 53 31 35 30 30 36 30 31 31 37 34 30 32
37 30 03 06 05 02 
aa 55 01 00 00 00 10 01 11 42 53 31 35 30 30 36 30 31 31 37 34 30 32 37 30 f8 05 71
aa 55 00 f8 01 00 10 81 01 06 02 90 
aa 55 01 00 00 f8 11 03 00 02 0c aa 55 00 f8 01 00 11 83 93 31 20
20 32 30 30 30 45 31 2e 30 30 20 20 20 20 5a 4c 20
31 35 30 30 53 20 20 20 20 20 20 20 20 5a 45 56 45
52 53 6f 6c 61 72 20 20 42 53 31 35 30 30 36 30 31
31 37 34 30 32 37 30 33 36 30 30 56 36 31 30 2d 30
31 30 33 37 2d 30 39 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
56 36 31 30 2d 30 31 30 33 38 2d 30 39 56 36 31 30
2d 31 30 30 30 30 2d 30 33 18 94 aa 55 01 00 00 f8
11`

The aa 55 lines is the standard header, so it represents good communication
aa 55 00 xx is Access point
aa 55 xx 03 is inverter 03

So 2 bytes header, 2 bytes source address 2 bytes destination address, 

It's the 7th and 9th byte that are interesting for communication:
7th is a control code there are 4 kinds:
'1. Register(0x10)
2. Read(0x11)
3. Write(0x12)
4. Execute(0x13)'

8th is a function code, 
9th is the data length, then comes the data

So to your data:
aa 55 01 00 00 00 10 00 00 01 10 
looks like a registration:

Control Code :0x10 ‘register’
Control code Function code Vector Description
0x10 0x00 AP -> Inverter Register query
0x10 0x80 Inverter-> AP registration request
0x10 0x01 AP -> Inverter send register address
0x10 0x81 Inverter-> AP address confirm
0x10 0x02 AP -> Inverter remove register
0x10 0x82 Inverter-> AP remove confirm
0x10 0x03 AP -> Inverter re-connect removed Inverter
0x10 0x04 AP -> Inverter Re-register

You'll see the 00 80 registration request in your data. So you have a system, that appears to communicate, and communicate slightly garbled.

It looks like you have more interters talking at the same time. Are you sure, the code runs only once?

It's either that, or a slightly different protocol as the data is shown now

BR

Henrik

blickdicht, 2017-10-19:
Hej Henrik
nope: only one inverter connected to one raspberry. 
I did another dump this afternoon (maybe with changed adress of the inverter, since i fiddled around quite a bit):
I cleaned it up with your help from above (thanks) and put it in a google spreadsheet: [goo.gl/NDstQj](url)

My new, but weak theory is:
The inverter is sold in a standard version with RS485. I bought the version with extra WiFi. It has a piggyback module with Ethernet and WiFi. Maybe, the piggyback module communicates with the baseboard via RS485. Therefore there would already be an AP in the network. The communication from raspi would disturb the already established comm between piggyback and baseboard.
A fact speaking against this theory is, that I can see no busactivity via LEDs on the RS485 adapter if the eversolar-monitor is not started. At least the RX LED should blink...
I will have a logic analyzer for the weekend - I'll be back with more details.
Georg

blickdicht, 2017-10-20:
Jep! Found it. You are right: eversolar-monitor was not alone on the bus.
Connecting the inverter via the RS485/USB FTDI Dongle to a PC and logging data to file with only a terminal program running gives:

  | H | H | S | S | D | D | CC | FC | L |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
  | AA | 55 | 01 | 00 | 00 | 00 | 10 | 00 | 00 | 01 | 10 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
  | AA | 55 | 01 | 00 | 00 | 00 | 10 | 00 | 00 | 01 | 10 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
Question | AA | 55 | 01 | 00 | 00 | C9 | 02 | 00 | 01 | DC |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
Answer | AA | 55 | 00 | C9 | 01 | 00 | 82 | 32 | 00 | AC | 00 | 15 | 04 | B2 | 00 | 00 | 00 | 0F | 00 | 00 | 00 | 09 | 08 | FE | 88 | 00 | A3 | FF | FF | 00 | 00 | 01 | 76 | 00 | 00 | 00 | A1 | 00 | 01 | 00 | 00 | 00 | 00 | FF | FF | 00 | 00 | 00 | 00 | 00 | 00 | 00 | 00 | FF | 00 | 00 | 00 | 0C | 75
repeat 7 x | AA | 55 | 01 | 00 | 00 | 00 | 10 | 00 | 00 | 01 | 10 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
Question to write sth. | AA | 55 | 01 | 00 | 00 | C9 | 12 | 02 | 04 | C0 | A8 | B2 | 38 | 04 | 33 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
Answer | AA | 55 | 00 | C9 | 01 | 00 | 12 | 82 | 01 | 06 | 02 | 64 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
Question to write sth. | AA | 55 | 01 | 00 | 00 | C9 | 12 | 03 | 01 | 01 | 01 | E0 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
Answer | AA | 55 | 00 | C9 | 01 | 00 | 12 | 83 | 01 | 06 | 02 | 65 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
  | AA | 55 | 01 | 00 | 00 | 00 | 10 | 00 | 00 | 01 | 10 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
  | AA | 55 | 01 | 00 | 00 | 00 | 10 | 00 | 00 | 01 | 10 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
Question | AA | 55 | 01 | 00 | 00 | C9 | 02 | 00 | 01 | DC |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  
Answer | AA | 55 | 00 | C9 | 01 | 00 | 82 | 32 | 00 | A9 | 00 | 15 | 04 | EE | 00 | 00 | 00 | 0E | 00 | 00 | 00 | 09 | 08 | FE | 88 | 00 | A1 | FF | FF | 00 | 00 | 01 | 76 | 00 | 00 | 00 | A1 | 00 | 01 | 00 | 00 | 00 | 00 | FF | FF | 00 | 00 | 00 | 00 | 00 | 00 | 00 | 00 | FF | 00 | 00 | 00 | 0C | AB
repeat 7 x | AA | 55 | 01 | 00 | 00 | 00 | 10 | 00 | 00 | 01 | 10 |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |   |  

Full data can be found in the second tab of the google spreadsheet linked above. It just continues like this...

So the WiFi-Module communicates via RS 485 with the baseboard. The company obviously uses their external comm-box sold seperately also as an internal WiFi module. 
It seems to give a new ID to the baseboard every day, when it registers first in the morning.

So how can I continue? My final targets would be, to 
- get the detailed data into domoticz and a local webserver without much delay 
- have a cloud-based longterm memory (which is already kind of provided by the manufacturer in china).
 
Do you think a new special "spy"-mode in eversolar-monitor would be a clean solution in listening to the already established communication? 
Do the protocols seem compatible to you at a first glance?
Is there any official documentation for the protocol?
I have no idea of perl programming, but with quite some C background, perl does not look too scary to me.
BR
Georg

solmoller, 2017-10-21:
Hi Georg
Great!

I don't know the WiFi module, but I'd guess you may be able to talk to that module with the Eversolar monitor perl code. So there are 4 pathways forward:

1. communicate through the RS485 and avoid clashes, that may be possible, if we lower the interactions per minute
2 Talk to the inverter through the WiFi interface with the perl script
3 Take the data from the cloud to Docomoticz
4 Spying is also an option, listening in and awaiting the AA55 signal, and then parsing the rest of the string. 

Yes it appears to be compatible
Yes we have the documentation, but we are not allowed to share it openly. PM me with your email. (Zeversolar has been extremely supportive when we set this up)

If you know C, then Perl should be easy

BR

Henrik

blickdicht, 2017-10-24:
Hi Henrik,
thanks again for your help. 
I fiddeld around with those options:

> 1.  communicate through the RS485 and avoid clashes, that may be possible, if we lower the interactions per minute 

If I read packet with length 256, I can not be sure, that I will not get only half of a packet, since I might have started to listen while communication is already going on. So I dump this first reception and listen immeadiately again for 256 bytes. The next packet will be either one complete packet, sent by the AP or a sent packet of the AP AND its response by the inverter.
I can even send packets to the inverter myself, ask for data and get a response, if I send my packet immediately after the AP/inverter was talking. The bus seems to be free for a certain amount of loop time. 
But once the inverter is registered with its own access point, it is deaf to any other registration packets from eversolar-monitor.

> 2 Talk to the inverter through the WiFi interface with the perl script

Well -  not that I know of. On the ETH/WiFi Interface there there is only the integrated webserver where i can scrape data from.

> 3 Take the data from the cloud to Docomoticz

Too slow. It only updates every 15 Min. I want to do controlled dumping of surplus energy. I have to be faster.

> 4 Spying is also an option, listening in and awaiting the AA55 signal, and then parsing the rest of the string. 

Thats what I try I at the moment. 
It works, but the whole software structure does not fit any more. I can not go for
`foreach inverter { ...}`
but rather 
`receive packet, 
analyze,
if ok then save into database entry with the right ID`

So at the moment I think, I will get it to work, but a commit to eversolar-monitor does not make sense.  The different structural needs of both solutions create lots of messy workarounds.

Furthermore the answer packet for "NORMAL_INFO"  with 0x11 / 0x02 code is not compatible at all.
Example:

My findings |   |   |   |   |   |   |   |   |   |   | Temp |   |   |   | Upv |   |   |   |   |   |   |   |   |   | Vac |   | F_ac |   |   |   |   |   |   |   | Etot |   |   |   | h? |   |  
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
Eversolar-monitor |   |   |   |   |   |   |   |   |   |   | E_TODAY |   | VPV |   | IPV |   | IAC |   | VAC |   | FREQ. |   | PAC |   | IMP. |   | NA_1 |   | E_TOT. |   | NA_2 |   | HOURS_UP |   | OP_MODE |   |   |   |   |   |  
Bytes | AA | 55 | 00 | 30 | 01 | 00 | 11 | 82 | 32 | 00 | 9A | 00 | 00 | 03 | 7F | 00 | 00 | 00 | 00 | 00 | 00 | 00 | 00 | 09 | 00 | 13 | 88 | 00 | 00 | FF | FF | 00 | 00 | 01 | A6 | 00 | 00 | 00 | B2 | 00 | 00
Value |   |   |   |   |   |   |   |   |   |   | 15,4 |   | 0,0 |   | 89,5 |   | 0,0 |   | 0,0 |   | 0,0 |   | 0,0 |   | 230,4 |   | 50,0 |   | 0,0 |   | 655,4 |   | 0,0 | 0,0 | 42,2 |   | 0,0 | 0,0 | 178,0 |   |  
Bytes | AA | 55 | 00 | 30 | 01 | 00 | 11 | 82 | 32 | 00 | C0 | 00 | 0B | 04 | B2 | 00 | 00 | 00 | 0C | 00 | 00 | 00 | 08 | 09 | 02 | 13 | 86 | 00 | 8A | FF | FF | 00 | 00 | 01 | A7 | 00 | 00 | 00 | B3 | 00 | 01
Value |   |   |   |   |   |   |   |   |   |   | 19,2 |   | 0,1 |   | 120,2 |   | 0,0 |   | 12,0 |   | 0,0 |   | 8,0 |   | 230,6 |   | 50,0 |   | 1,4 |   | 655,4 |   | 0,0 | 0,0 | 42,3 |   | 0,0 | 0,0 | 179,0 |   |  

Another indication for this incompatibility is an experiment I made:
As I wrote above, the Ethernet/WiFi Module is a piggyback module merely connected by one small ribbon cable. Disconnecting this cable and starting the inverter makes it run normally without the webserver and the Ethernet Interface obviously. 
On the now free RS 485 there is no communication anymore. Fine!
Starting Eversolar-monitor works, but I get frequent communication problems. There are changing values in the webserver, but they are all wrong. Btw: the checksum is not implemented in the ZL 1500 and gives 0x01 0x10 always.
--> the protocol is closely related, but not compatible.

So knowing all this, I will 

1. have a look at some simple webserver-scraping-solutions
2. continue on a variant in spy mode, but give up the compatibility to eversolar-monitor

If I arrive anywhere interesting, I'll post it here.

I would really like to have a look into the specification, but either there is no PM possibility in Git anymore or I do not find it. Anyway: here is a temporary email for getting in contact with me: wypuj@nada.ltd
 

Regards, georg

solmoller, 2017-10-26:
PM sent

Regards

Henrik

Steve9R, 2018-01-20:
Just a comment.. 

I am running an Zeversolar TL-5000 successfully with eversolar-monitor.. but im not using wifi, i ran a cat 5 cable to the RS-485 and use an ATC-1000 RS-485 to ethernet converter into my Raspberry Pi running eversolar-monitor.. 

i tried a few cheaper converters and had no luck until i got the ATC-1000 (ebay) 

Been working fine since 2015..

solmoller, 2018-01-21:
Hi Steve

I have added the ATC-1000 to known good hardware, thanks!

Henrik

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virusfri.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

phungorquan, 2019-08-27:
Hello everybody, i was stuck with 3000tl zeversolar, when i connected the usb-rs485 to rj45 of inverter, i didn't receive anything. I tried to send some command afterthat but i also didn't receive.

I buy this in my country (VietNam) : https://thegioiic.com/products/mach-chuyen-doi-usb-to-rs485

I don't know why the inverter doesn't response anything although i tried to send many commands continously. I used hercules terminal to read serial from the usb.

I tried to commute 2 usb-rs485 each other, it can normally send and receive.The inverter is on grid and operate normally

This is the family zeversolar inverter document: https://drive.google.com/file/d/1kzXyG2114c4oZBB9dWruK63_X4J58sIg/view?usp=drivesdk

I followed step by step , re-register by send the command x3 , then send query ,... but i can not receive anything

It does's have any led on the inverter side, on the usb-rs485 i saw the tx led was blink.
When i plug the rs485 into inverter for the first time, it sent some noise signals to my computer, that's mean my wires aren't broken

phungorquan, 2019-08-29:
Thanks too much, it's work

closed 2019-08-29 by solmoller