solmoller/eversolar-monitor #26: Webserver doesn't show output from both inverters

jessgade, 2017-12-26:
I've downloaded version 0.17 but the webserver only shows the output from one inverter.
I'm using a Raspberry Pi Zero W and have two Eversolar TL-3000 inverters.
The Combined Energy is also only the contribution from the same inverter and not both.
I have the same problem with uploads to PVOutput - but I dont know if its related.

[image]

I examined the Perl code (I have no experience in perl) and made a slight change, since I've noticed that a set of variables are declared but not really used.
I changed only the second last line in the following code from :

```
if($response) {
# good response - reset response_timeout_count
$inverters{$inverter}{"response_timeout_count"} = 0;

my $len = length($response);
my @data = parse_packet(unpack("C$len", $response));

my $e_today_kwh = $data[$DATA_BYTES{'E_TODAY'}]/100;
my $e_today_wh = $e_today_kwh * 1000;
my $e_total = $data[$DATA_BYTES{'E_TOTAL'}]/10 +$data[$DATA_BYTES{'NA_1'}]*65535/10; # NA_1 is actually upper bytes of total production
my $pac = $data[$DATA_BYTES{'PAC'}];
if ($data[$DATA_BYTES{'TEMP'}]>=0x8000) {$data[$DATA_BYTES{'TEMP'}]-=0x10000;} # Temperature is signed, -0.1 = 0xFFFF
my $temp = $data[$DATA_BYTES{'TEMP'}]/10;
my $volts = $data[$DATA_BYTES{'VAC'}]/10;

$combined_power = $combined_power +$data[$DATA_BYTES{'PAC'}];
$combined_daykwh = $combined_daykwh +$data[$DATA_BYTES{'E_TODAY'}]/100;

```
**to:**

```
if($response) {
# good response - reset response_timeout_count
$inverters{$inverter}{"response_timeout_count"} = 0;

my $len = length($response);
my @data = parse_packet(unpack("C$len", $response));

my $e_today_kwh = $data[$DATA_BYTES{'E_TODAY'}]/100;
my $e_today_wh = $e_today_kwh * 1000;
my $e_total = $data[$DATA_BYTES{'E_TOTAL'}]/10 +$data[$DATA_BYTES{'NA_1'}]*65535/10; # NA_1 is actually upper bytes of total production
my $pac = $data[$DATA_BYTES{'PAC'}];
if ($data[$DATA_BYTES{'TEMP'}]>=0x8000) {$data[$DATA_BYTES{'TEMP'}]-=0x10000;} # Temperature is signed, -0.1 = 0xFFFF
my $temp = $data[$DATA_BYTES{'TEMP'}]/10;
my $volts = $data[$DATA_BYTES{'VAC'}]/10;

$combined_power = $combined_power + $pac;
$combined_daykwh = $combined_daykwh +$data[$DATA_BYTES{'E_TODAY'}]/100;

```

And to my surprise  using the $pac variable made the webserver output work?!
The Combined energy also suddenly work despite the fact that the declared variable for this haven't been used.
The output to PVOutput still doesent work though - its still only the one inverter output thats registered.

solmoller, 2017-12-28:
Well this is strange! I just started looking at the problem, and I cannot recreate it, but I saw the problem just a few days ago.

I think I know what is wrong though: If you have lost the connection to one inverter, then the comment in the code tells you what is wrong:
# request data from each connected inverter
     
As I read the code, it only sums up on connected inverters, so the code fails at the end of the day, when the first inverter stalls.

Incidentally, your 365 days production looks quite off on on at least one account: You have 7.376, 4.165 kWh in the top screenshot, and 13.930 / 10.718 kWh in the lower screenshot. _If you upgraded the perl script between the two runs I understand, if you did not... I have no idea what is right, though I'd expect the lowest number to be right, but if you have deleted the SQL file between rund, then the system subtracts now with 365 days ago (zero), and you get the system production of all time. Did you delete the SQL file?

BR

Henrik

jessgade, 2017-12-28:
I've looked a bit further on this.
I had the problem reappear even with my little initial code change.

I'm convinced that the problem is that the JSON serialization/deserialization is non-deterministic 
A sorting mechanism needs to be applied to the perl JSON serialization and a similar sort needs to be in the HTML reading mechanism to ensure that the data is read in the same order.

This would explain why a slight code change would affect the program execution to 'magically' make the data appear correct.

I made an ugly hack where I changed the HTML code to find the Max totalpower and Max totalkwh from the read data and this makes it work again.

solmoller, 2017-12-28:
..or the code has been run with one, respectively two inverters connected?

closed 2018-06-21 by solmoller

oz1paj, 2019-10-07:
This is still a problem, I has always been there.
[image]

Best Regards
Peter Lanng