solmoller/eversolar-monitor #27: 2 inverters but registering only data under one inverter id

jessgade, 2018-01-02:
I've recently starting with this fine piece of software but have (another problem):

I have two inverters;
Q0067 (been producing ~13939 kWh) and
P0028 (~10727 kWh).

But some days only the id for one inverter gets registered.  So far its allways P0028 wich 'wins'. The log below shows this. 
All data subsequently gets registered with the P0028 inverter id.
Other days it works fine - Anyone got an idea? :-)
-------------------------

2018-01-02 08:49:11: Severity 3, No response to 'offline query' request - no offline inverters
2018-01-02 08:49:13: Severity 3, No response to 'offline query' request - no offline inverters
2018-01-02 08:49:15: Severity 3, No response to 'offline query' request - no offline inverters
2018-01-02 08:49:17: Severity 3, No response to 'offline query' request - no offline inverters
2018-01-02 08:49:19: Severity 1, Found serial number: B983000A128Q0067
2018-01-02 08:49:21: Severity 1, Inverter acknowledged registration
2018-01-02 08:49:23: Severity 2, No response to 'query inverter id' request for inverter B983000A128Q0067
2018-01-02 08:49:25: Severity 3, No response to 'offline query' request - no offline inverters
2018-01-02 08:49:27: Severity 3, No response to 'offline query' request - no offline inverters
2018-01-02 08:49:29: Severity 3, No response to 'offline query' request - no offline inverters
2018-01-02 08:49:31: Severity 3, No response to 'offline query' request - no offline inverters
2018-01-02 08:49:33: Severity 3, No response to 'offline query' request - no offline inverters
2018-01-02 08:49:35: Severity 3, No response to 'offline query' request - no offline inverters
2018-01-02 08:49:37: Severity 1, Found serial number: B983000A128P0028
2018-01-02 08:49:39: Severity 2, No response to 'send register address' request for inverter B983000A128P0028
2018-01-02 08:49:41: Severity 3, No response to 'offline query' request - no offline inverters
2018-01-02 08:49:43: Severity 3, No response to 'offline query' request - no offline inverters
2018-01-02 08:49:45: Severity 3, No response to 'offline query' request - no offline inverters
2018-01-02 08:49:47: Severity 3, No response to 'offline query' request - no offline inverters
2018-01-02 08:49:49: Severity 3, No response to 'offline query' request - no offline inverters
2018-01-02 08:49:51: Severity 3, No response to 'offline query' request - no offline inverters
2018-01-02 08:49:53: Severity 1, Found serial number: B983000A128P0028
2018-01-02 08:49:55: Severity 1, Inverter acknowledged registration
2018-01-02 08:49:57: Severity 2, No response to 'query inverter id' request for inverter B983000A128P0028
2018-01-02 08:49:59: Severity 3, No response to 'offline query' request - no offline inverters
2018-01-02 08:50:01: Severity 3, No response to 'offline query' request - no offline inverters
2018-01-02 08:50:03: Severity 3, No response to 'offline query' request - no offline inverters
2018-01-02 08:50:05: Severity 3, No response to 'offline query' request - no offline inverters
2018-01-02 08:50:07: Severity 1, Found serial number: B983000A128P0028
2018-01-02 08:50:09: Severity 1, Inverter acknowledged registration
2018-01-02 08:50:11: Severity 2, No response to 'query inverter id' request for inverter B983000A128P0028
2018-01-02 08:50:13: Severity 3, No response to 'offline query' request - no offline inverters
2018-01-02 08:50:15: Severity 3, No response to 'offline query' request - no offline inverters
2018-01-02 08:50:17: Severity 3, No response to 'offline query' request - no offline inverters
2018-01-02 08:50:19: Severity 3, No response to 'offline query' request - no offline inverters
2018-01-02 08:50:21: Severity 1, Found serial number: B983000A128P0028
2018-01-02 08:50:23: Severity 1, Inverter acknowledged registration
2018-01-02 08:50:25: Severity 1, Connected to inverter: 1  3000E1.00      TL3000                    B983000A128P00283600V610-01011-06
2018-01-02 08:50:25: Severity 2, Asking for any inverters to register
2018-01-02 08:50:27: Severity 3, No response to 'offline query' request - no offline inverters
2018-01-02 08:50:29: Severity 2, B983000A128P0028 lost contact with inverter (1 time(s))
2018-01-02 08:50:42: Severity 3, B983000A128P0028 output: 0 W, Total: 13939 kWh, Today: 0 kWh, 365 days : 13939 
2018-01-02 08:50:54: Severity 3, B983000A128P0028 output: 0 W, Total: 13939 kWh, Today: 0 kWh, 365 days : 13939 
2018-01-02 08:51:04: Severity 2, Asking for any inverters to register
2018-01-02 08:51:06: Severity 3, No response to 'offline query' request - no offline inverters
2018-01-02 08:51:08: Severity 3, B983000A128P0028 output: 0 W, Total: 13939 kWh, Today: 0 kWh, 365 days : 13939 
2018-01-02 08:51:21: Severity 3, B983000A128P0028 output: 0 W, Total: 13939 kWh, Today: 0 kWh, 365 days : 13939 
2018-01-02 08:51:33: Severity 3, B983000A128P0028 output: 0 W, Total: 13939 kWh, Today: 0 kWh, 365 days : 13939 
2018-01-02 08:51:45: Severity 3, B983000A128P0028 output: 0 W, Total: 13939 kWh, Today: 0 kWh, 365 days : 13939 
2018-01-02 08:51:57: Severity 3, B983000A128P0028 output: 0 W, Total: 13939 kWh, Today: 0 kWh, 365 days : 13939 
2018-01-02 08:52:08: Severity 2, Asking for any inverters to register
2018-01-02 08:52:10: Severity 1, Found serial number: B983000A128P0028
2018-01-02 08:52:12: Severity 1, Inverter acknowledged registration
2018-01-02 08:52:14: Severity 1, Connected to inverter: 1  3000E1.00      TL3000                    B983000A128P00283600V610-01011-06
2018-01-02 08:52:16: Severity 3, B983000A128P0028 output: 0 W, Total: 13939 kWh, Today: 0 kWh, 365 days : 13939 
2018-01-02 08:52:18: Severity 3, B983000A128P0028 output: 0 W, Total: 10727.4 kWh, Today: 0 kWh, 365 days : 10727.4 
2018-01-02 08:52:31: Severity 3, B983000A128P0028 output: 0 W, Total: 13939 kWh, Today: 0 kWh, 365 days : 13939 
2018-01-02 08:52:33: Severity 3, B983000A128P0028 output: 0 W, Total: 10727.4 kWh, Today: 0 kWh, 365 days : 10727.4 
2018-01-02 08:52:45: Severity 3, B983000A128P0028 output: 0 W, Total: 13939 kWh, Today: 0 kWh, 365 days : 13939 
2018-01-02 08:52:48: Severity 3, B983000A128P0028 output: 0 W, Total: 10727.4 kWh, Today: 0 kWh, 365 days : 10727.4 
2018-01-02 08:53:00: Severity 3, B983000A128P0028 output: 0 W, Total: 13939 kWh, Today: 0 kWh, 365 days : 13939 
2018-01-02 08:53:02: Severity 3, B983000A128P0028 output: 0 W, Total: 10727.4 kWh, Today: 0 kWh, 365 days : 10727.4 
2018-01-02 08:53:12: Severity 2, Asking for any inverters to register
2018-01-02 08:53:14: Severity 3, No response to 'offline query' request - no offline inverters
2018-01-02 08:53:17: Severity 3, B983000A128P0028 output: 0 W, Total: 13939 kWh, Today: 0 kWh, 365 days : 13939 
2018-01-02 08:53:19: Severity 3, B983000A128P0028 output: 0 W, Total: 10727.4 kWh, Today: 0 kWh, 365 days : 10727.4 
2018-01-02 08:53:31: Severity 3, B983000A128P0028 output: 0 W, Total: 13939 kWh, Today: 0 kWh, 365 days : 13939 
2018-01-02 08:53:33: Severity 3, B983000A128P0028 output: 0 W, Total: 10727.4 kWh, Today: 0 kWh, 365 days : 10727.4 
2018-01-02 08:53:46: Severity 3, B983000A128P0028 output: 0 W, Total: 13939 kWh, Today: 0 kWh, 365 days : 13939 
2018-01-02 08:53:48: Severity 3, B983000A128P0028 output: 0 W, Total: 10727.4 kWh, Today: 0 kWh, 365 days : 10727.4 
2018-01-02 08:54:00: Severity 3, B983000A128P0028 output: 0 W, Total: 13939 kWh, Today: 0 kWh, 365 days : 13939 
2018-01-02 08:54:03: Severity 3, B983000A128P0028 output: 0 W, Total: 10727.4 kWh, Today: 0 kWh, 365 days : 10727.4

solmoller, 2018-01-02:
We have had some crosstalk before. I can probably dig it out from
'solcelledebat'.

Meanwhile I suggest you disable 'agressive polling'

Br Henrik

jessgade, 2018-01-03:
Thank you for the response. 
I have disabled agressive polling and will see if that does a difference.

closed 2018-06-21 by solmoller