solmoller/eversolar-monitor #28: Re-upload old data ?

Steve9R, 2018-01-20:
Hi guys, 
Ive been using Eversolar-monitor on our TL-5000 now since November 2015 uploading to PVoutput.
(yes im a member of the Eversolar Team)

But.. back in december I changed ISP's and of course DNS failed and i didnt realise, i fixed it on the Pi and its uploading again, but for that entire period from 13/12 -> today (21/01) there is no data in pvoutput, yet my system has been collecting and adding it into the sqlite db.. 

is there a way of extracting and uploading the missing days from the db back to pvoutput ? 

Steve

solmoller, 2018-01-21:
Hi Steve

You can export daily data as described here:

https://github.com/solmoller/eversolar-monitor/blob/wiki/Usage.md

It should be easy to upload or type in the data in PVoutput.

BR

Henrik

solmoller, 2018-01-21:
As you're not the first to miss out on the extraction, I noticed, that there was no link to the relevant wiki page. Fixed!

closed 2018-06-21 by solmoller