solmoller/eversolar-monitor #31: Wifi not working

Tiko12, 2018-06-21:
Hi 
Tx. for at fantastic project - and good description

I am building a Ras pi (B) - but are having the problem that it seems that the wifi dongle is not activated/enabled.
The same Ras pi - works well with KODI (medial player) - so I guess that the Ras and wifi dongle (USB) is ok

How can I get it ti work
Has just plugged in LAN cable and get IP a.o.
Was hoping to be able to get update or whatever is needed to get it ti work

Thank you in advance
Br. Tiko12

solmoller, 2018-06-21:
Hi Tiko,

I doubt there is a problem with the WIFI dongle, if you can use the RasPi. I also doubt there is a problem with the software. I need a more detailed description on how you see the problem:

When you use the eversolar-image, can you browse the web?
Maybe it is a newer dongle that is not supported by the ancient software stack, have you updated the operating system and raspi?

I suggest you take a path, that focuses on getting the dongle to work in general, and then look at the Eversolar software afterwards.

BR Henrik

Tiko12, 2018-06-22:
Hej Henrik

Jeg er ikke så skrap til Rasp - så jeg håbede på et image der var 'klar' :-)

Har ikke hentet et basic Rasp image - men taget det der var på siden.
Forsøgte at opdatere - apt-get update
men den brokker sog over skrivebeskytelse og foreslå at jeg logger ind med
root

Har ikke kodeord til root

Håber en opdatering kan gøre det.

HW virker for med at færdig image fra ex. KODE virker...

I wifi konfiguratoren dukker der ikke noget kort op....

På forhånd mange tak
mvh. Steen (aka Tiko12)

solmoller, 2018-06-25:
Det lyder som om der er noget galt med det image. Prøv at downloade igen, og se, om det hjælper. sudo apt-get update bør ikke spørge efter root pwd.

MVH

Henrik

Tiko12, 2018-06-25:
Hej

Tak for svar .-)
Har kørt sudo apt-get update en stribe gange nu.

Vil dog fortsat ikke melde noget wifi (adapter).

Underligt da samme HW virker med andet image..... sidder og tumler med det
nu.. :-)

mvh Steen

Tiko12, 2018-07-20:
Hentede en std. Debian - og alt virker.
Hvordan få jeg så EverSolar på ...?

Kan der evt. laves et opdateret image -i håb om at mit wifi så er supporteret. Altså ready2go image med EverSolar ? :-)
Har netop modtaget RS485/USB adapter - så hvis jeg kunne få det til at virker - så er alt HW klar :-)

solmoller, 2018-08-03:
Der er instruktioner til installation på sådan en platform her:

https://github.com/solmoller/eversolar-monitor/blob/wiki/Software_installation.md

MVH

Henrik

closed 2021-07-21 by solmoller