solmoller/eversolar-monitor #32: add tag to the webserver inverter.data.pac to enable interaction with other devices i.e. esp8266's

whereswally606, 2018-07-17:
change from
on the webserver page.
<div style="color: darkgreen; font-size: 36pt; font-weight: bold; float: right;">'+inverter.data.pac+'<span style="font-size: 14pt;">W</span></div>

to 

<div class="inverter_pac" style="color: darkgreen; font-size: 36pt; font-weight: bold; float: right;">'+inverter.data.pac+'<span style="font-size: 14pt;">W</span></div>

so that it is possible in one line using a webscraping esp8266 to get the continuous power.
Currently I have to

var pac = document.getElementsByClassName("inverter_content")[0].innerHTML.substr(93)
pac.slice(0, pac.indexOf("<") )

this will help me make a homebrew charge limiter to collect only the excess solar energy into our Nissan leaf. Either using openevse or actually spoofing the pilot signal on our podpoint charge point.

Thanks for your help. I tried to see where this was being done but it want obvious to me that it was in the .pl file at all.

Currently 5 days in and have just joined eversolar perl loggers on pvoutput.org

solmoller, 2018-07-17:
> '+inverter.data.pac+'W

Interesting!

I am unsure I understand the change request, but data can be presented as
you wish. Simplest way is getting latest data from mysql, that should be
easy.

(I am traveling and your change request is presented as two identical lines)

BR Henrik

whereswally606, 2018-07-17:
no problem I actually solved the issue with 
document.getElementsByClassName("inverter_content")[0].childNodes.item(1).childNodes.item(1).childNodes.item(0)
which get me the power ac in one line so I shouldn't need the tag now. Im sorry for prematurely raising and issue.

solmoller, 2018-07-18:
Good to hear

Can you please describe your solution?  What do you use the esp8266 for,
and how is it set up?

BR Henrik

closed 2021-05-06 by solmoller