solmoller/eversolar-monitor #34: can't locate object method" new" via package " Device::Serialport"

emichiels, 2018-08-29:
Hi there,

I'm trying to use this software on a raspberry pi to read out my eversolar tl300. I tried to install with the image, but the image didn't run. Then I installed raspbian and added eversolar-monitor conform the installation procedure. Now it keeps saying . Can't locate object" new"  via package "Device::Serialport"  (perhaps you forgot to load " Device::Serialport"?) at eversolar.pl line 503.
What am I doing wrong? 
I use a standard usb to rs485 converter bought on aliexpress. with A and B connected to pin 1 and 2 and GND connected to pin 4. the 5V not connected.  Is this the right way to do it?

solmoller, 2018-08-29:
I need a little more to go with here:

The image did not run. As in didn't boot ? If you have similar problems with the image and your own installation, then there might be a problem with the serial port.

If you are swapping images, consider doing a backup or another SD card - as you know by now, the installation from scratch is a lengthy one.

Anyway, googling your error leads me to believe, that you have omitted this bit:

> apt-get install libdevice-serialport-perl

The cabling does sound in line with https://github.com/solmoller/eversolar-monitor/blob/wiki/Hardware.md - but right now you have a software error to fix.

Good luck!

Henrik

emichiels, 2018-08-29:
Hi Henrik, 

Thanks for your fast reaction. 
The image didn't run because it did not boot. I tried 2 different times a download of the image. but both did not make the Raspberry Pi boot.  So therefor I don't know if the image would give the same problem.

I added the libdevice-serialport-perl  at that seemed to be the problem. The software problem is now solved. 

But now I have the hardware problem. I've bough a USB to RS485 converter like this one:
https://www.aliexpress.com/item/NEW-Industrial-USB-To-RS485-Converter-Upgrade-Protection-RS485-Converter/32638090708.html?spm=2114.search0104.3.1.93391b70yaHt7b&ws_ab_test=searchweb0_0,searchweb201602_3_10065_10068_5723115_5910015_5723715_5890015_5723815_10696_5723215_5724015_5723415_10084_10083_5970015_10618_5724315_10304_10307_10820_10301_10821_5724215_5723315_5724115_5723615_10843_10059_5723015_100031_5723915_5725015_10103_5725115_5980015_5724915_5990015,searchweb201603_54,ppcSwitch_5&algo_expid=50c5748b-d9a4-4a56-b40f-f88b9ccc64cc-0&algo_pvid=50c5748b-d9a4-4a56-b40f-f88b9ccc64cc&priceBeautifyAB=0 

As you can see it has A,B,GND and +5V . 
I have been looking at the hardware page, but couldn't figure out the mapping. 
Am I using a wrong kind of RS 485 converter? 

Kind regards,

Erik Michiels

solmoller, 2018-08-29:
Now that is a cheap adapter!  Try logging everything, and see what it says

emichiels, 2018-08-30:
this is what's being logged: 

_Update old database version, or print 2 fail messages :
DBD::SQLite::db do failed: duplicate column name: vpv2 at eversolar.pl line 799.
DBD::SQLite::db do failed: duplicate column name: ipv2 at eversolar.pl line 800.
Done updating old database version.
2018-08-30 13:08:07: Severity 3, Checking database
SolarWebServer: You can connect to your server at http://localhost:3837/
2018-08-30 13:08:07: Severity 2, Connecting to the serial port
2018-08-30 13:08:07: Severity 2, Asking all inverters to re-register
sending packet to inverter...
aa 55 01 00 00 00 10 04 00 01 14
sending packet to inverter...
aa 55 01 00 00 00 10 04 00 01 14
sending packet to inverter...
aa 55 01 00 00 00 10 04 00 01 14
sending packet to inverter...
aa 55 01 00 00 00 10 04 00 01 14
sending packet to inverter...
aa 55 01 00 00 00 10 04 00 01 14
sending packet to inverter...
aa 55 01 00 00 00 10 04 00 01 14
sending packet to inverter...
aa 55 01 00 00 00 10 04 00 01 14
sending packet to inverter...
aa 55 01 00 00 00 10 04 00 01 14
sending packet to inverter...
aa 55 01 00 00 00 10 00 00 01 10
2018-08-30 13:08:27: Severity 3, No response to 'offline query' request - no offline inverters
sending packet to inverter...
aa 55 01 00 00 00 10 00 00 01 10
2018-08-30 13:08:29: Severity 3, No response to 'offline query' request - no offline inverters
sending packet to inverter...
aa 55 01 00 00 00 10 00 00 01 10
2018-08-30 13:08:31: Severity 3, No response to 'offline query' request - no offline inverters
sending packet to inverter...
aa 55 01 00 00 00 10 00 00 01 10
2018-08-30 13:08:33: Severity 3, No response to 'offline query' request - no offline inverters
sending packet to inverter...
aa 55 01 00 00 00 10 00 00 01 10
2018-08-30 13:08:35: Severity 3, No response to 'offline query' request - no offline inverters
sending packet to inverter...
aa 55 01 00 00 00 10 00 00 01 10_

So looks like all is working until the USB-RS485 converter. According to me it's a problem that the converter is half duplex and all documentation on the hardware page, for the cabling is for full duplex. So I'm trying to find out how to set up the cable for half duplex.

solmoller, 2018-08-30:
ok, you have progress

Out of couriousity, are the lights on the serial port flashing? Are there
any lights at the inverter side, that indicates incoming signals?

BR

Henrik

emichiels, 2018-08-31:
hi henrik,

Yes indeed lights are flashing. It's all working. I have some wifi stability issues because of the distance, but I'm working on that.  The web page is showing the data (See attachted screenshot in my comments above: https://github.com/solmoller/eversolar-monitor/issues/34#issuecomment-417283676  )

Now I got the next challenge: I have a tl 3000 with 2 strings attached. In Eversolar.ini I said that it's got 2 strings. but stil it only gives one result.

solmoller, 2018-08-31:
Try setting it to one string...

closed 2020-02-23 by solmoller