solmoller/eversolar-monitor #35: d365 to influx for "Health of system"

sorentorp, 2018-09-05:
Hi
I would like to add the d365 parameter to my InfluxDB, so that I can present the "health of my system" panel in Grafana.

What parameter would I be taking? I cant seem to find out if you store it in the sqlite database, then I would just be able to reuse that.

BTW - great project :-)

solmoller, 2018-09-05:
Thanks

d365 is not in the database, but with a little SQL knowledge you can probably convert this gem to fit your needs:
```

  my $stmt = "select min(e_total) as mins from inverter where timestamp  >=   date('now', '-365 day') and serial_number = '".$inverters{$inverter}{"serial"}."'";
            my $sth = $dbh->prepare( $stmt );
            my $rv = $sth->execute() or die $DBI::errstr;
             if($rv < 0){
               print $DBI::errstr;
             }
            while(my @row = $sth->fetchrow_array()) {
                            print "etot = ". $row[0] . "\n";
            $d365 = $e_total-$row[0] ;
                }

```

d365 is actually a quite good observation point - my system is apparently at an all time high :-)

sorentorp, 2018-09-07:
I'm not that firm a coder - wouldn't it just be possible to call up "$d365"?
My line to influx today is 
`my $cmd = `curl -is -XPOST "http://192.168.1.6:8086/write?db=eversolar" --data-binary "inverter_Stats,Type=Energy,Interface=inverter PAC=$pac,EToday=$e_today_wh,ETotal=$e_total,Temp=$temp,VPV=$VPV,VPV2=$VPV2,IPV=$IPV2,IPV2=$IPV2,VAC=$VAC,IAC=$IAC,FREQ=$FREQUENCY"`;`

and it doesn't seem to add any data when I use

`my $cmd = `curl -is -XPOST "http://192.168.1.6:8086/write?db=eversolar" --data-binary "inverter_Stats,Type=Energy,Interface=inverter PAC=$pac,EToday=$e_today_wh,ETotal=$e_total,Temp=$temp,VPV=$VPV,VPV2=$VPV2,IPV=$IPV2,IPV2=$IPV2,VAC=$VAC,IAC=$IAC,FREQ=$FREQUENCY,D365=$D365"`;`

sorentorp, 2018-09-10:
Works like a charm - thanks :-)

[image]

closed 2018-09-11 by sorentorp