solmoller/eversolar-monitor #36: Update old database version, or print 2 fail messages

sorentorp, 2018-09-05:
Any ideas on how to get around this issue?

Update old database version, or print 2 fail messages :
DBD::SQLite::db do failed: duplicate column name: vpv2 at eversolar.pl line 708.
DBD::SQLite::db do failed: duplicate column name: ipv2 at eversolar.pl line 709.
Done updating old database version.

wotid, 2018-09-05:
You're stretching my memory, but I think I solved that myself by commenting out a couple of lines after 

## START OF PROGRAM

## Setup database

Mine looks like this:

# vpv2 and ipv2 were added in later versions - fix any existing databases that don't have the columns
# print "Update old database version, or print 2 fail messages : \n";
# $dbh->do("ALTER TABLE inverter ADD COLUMN vpv2 FLOAT") or 1;
# $dbh->do("ALTER TABLE inverter ADD COLUMN ipv2 FLOAT") or 1;
# print "Done updating old database version.  \n";

Regards,

Steve.

solmoller, 2018-09-05:
Is it an issue?

Update old database version, **or print 2 fail messages** :
DBD::SQLite::db do failed: duplicate column name: vpv2 at eversolar.pl line 708.
DBD::SQLite::db do failed: duplicate column name: ipv2 at eversolar.pl line 709.
Done updating old database version.

The old database is updated first time around, thereafter you get the two messages. I tried getting the code to behave nicely, but didn't figure it was worth the hasstle

sorentorp, 2018-09-06:
Ah okay. Thank you!

closed 2018-09-07 by sorentorp