solmoller/eversolar-monitor #39: Upload to PVOutput.otg

GPellemans, 2019-02-14:
Hello, I use PVOutput since last week. I want every 5 minutes an upload to PVOutput.org
After 6 uploads (half an hour), it looks as if PVOutput stops receiving uploads. After an hour the receiving starts again: 6 times. Then another hour of no activity, And so on.
When i use an interval of 10 minutes, it's the same story: 3 uploads, an hour rest, 3 uploads, and so on. 

When i look at the internet, it looks as if i am the only one with this problem.

Does anyone regonize this problem and knows how to help me? Is it eversolar.pl that stops uploading, or is it PVOutput that stops reveiving?
Thanks in advance.

Gerard

Below are the times op the received uploads from today:

14/02/19 | 5:30PM | 7.720kWh | 3.574kWh/kW | 110W | 120W | 0.056kW/kW | 31.0C | 233.1V | - | - |  
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
14/02/19 | 5:25PM | 7.710kWh | 3.569kWh/kW | 196W | 240W | 0.111kW/kW | 31.3C | 232.3V | - | - |  
14/02/19 | 5:20PM | 7.690kWh | 3.560kWh/kW | 246W | 240W | 0.111kW/kW | 31.6C | 233.1V | - | - |  
14/02/19 | 5:15PM | 7.670kWh | 3.551kWh/kW | 268W | 360W | 0.167kW/kW | 31.9C | 233.9V | - | - |  
14/02/19 | 5:10PM | 7.640kWh | 3.537kWh/kW | 308W | 360W | 0.167kW/kW | 32.4C | 233.9V | - | - |  
14/02/19 | 5:05PM | 7.610kWh | 3.523kWh/kW | 426W | 360W | 0.167kW/kW | 32.7C | 235.5V | - | - |  
14/02/19 | 5:00PM | 7.580kWh | 3.509kWh/kW | 439W | 630W | 0.292kW/kW | 32.8C | 233.9V | - | - |  
14/02/19 | 4:00PM | 6.950kWh | 3.218kWh/kW | 779W | 720W | 0.333kW/kW | 35.5C | 237.0V | - | - |  
14/02/19 | 3:55PM | 6.890kWh | 3.190kWh/kW | 894W | 960W | 0.444kW/kW | 35.9C | 237.0V | - | - |  
14/02/19 | 3:50PM | 6.810kWh | 3.153kWh/kW | 951W | 1,080W | 0.500kW/kW | 36.1C | 237.0V | - | - |  
14/02/19 | 3:45PM | 6.720kWh | 3.111kWh/kW | 966W | 960W | 0.444kW/kW | 36.2C | 237.0V | - | - |  
14/02/19 | 3:40PM | 6.640kWh | 3.074kWh/kW | 999W | 1,080W | 0.500kW/kW | 36.1C | 235.5V | - | - |  
14/02/19 | 3:35PM | 6.550kWh | 3.032kWh/kW | 965W | 960W | 0.444kW/kW | 36.1C | 236.3V | - | - |  
14/02/19 | 3:30PM | 6.470kWh | 2.995kWh/kW | 967W | 1,090W | 0.505kW/kW | 36.4C | 235.5V | - | - |  
14/02/19 | 2:25PM | 5.290kWh | 2.449kWh/kW | 1,282W | 1,320W | 0.611kW/kW | 37.9C | 241.0V | - | - |  
14/02/19 | 2:20PM | 5.180kWh | 2.398kWh/kW | 1,210W | 1,200W | 0.556kW/kW | 37.9C | 240.2V | - | - |  
14/02/19 | 2:15PM | 5.080kWh | 2.352kWh/kW | 1,152W | 1,200W | 0.556kW/kW | 38.0C | 241.0V | - | - |  
14/02/19 | 2:10PM | 4.980kWh | 2.306kWh/kW | 1,394W | 1,320W | 0.611kW/kW | 38.3C | 241.0V | - | - |  
14/02/19 | 2:05PM | 4.870kWh | 2.255kWh/kW | 1,260W | 1,320W | 0.611kW/kW | 38.2C | 241.0V | - | - |  
14/02/19 | 2:00PM | 4.760kWh | 2.204kWh/kW | 1,313W | 1,320W | 0.611kW/kW | 38.5C | 241.8V | - | - |  
14/02/19 | 1:55PM | 4.650kWh | 2.153kWh/kW | 1,430W | 1,340W | 0.620kW/kW | 38.5C | 241.0V | - | - |  
14/02/19 | 12:55PM | 3.310kWh | 1.532kWh/kW | 1,317W | 1,200W | 0.556kW/kW | 36.2C | 240.2V | - | - |  
14/02/19 | 12:50PM | 3.210kWh | 1.486kWh/kW | 1,077W | 1,440W | 0.667kW/kW | 36.2C | 240.2V | - | - |  
14/02/19 | 12:45PM | 3.090kWh | 1.431kWh/kW | 1,459W | 1,440W | 0.667kW/kW | 35.9C | 241.8V | - | - |  
14/02/19 | 12:40PM | 2.970kWh | 1.375kWh/kW | 1,452W | 1,560W | 0.722kW/kW | 35.5C | 241.8V | - | - |  
14/02/19 | 12:35PM | 2.840kWh | 1.315kWh/kW | 1,376W | 1,200W | 0.556kW/kW | 35.0C | 241.0V | - | - |  
14/02/19 | 12:30PM | 2.740kWh | 1.269kWh/kW | 919W | 840W | 0.389kW/kW | 34.6C | 239.4V | - | - |  
14/02/19 | 12:25PM | 2.670kWh | 1.236kWh/kW | 1,352W | 1,230W | 0.569kW/kW | 35.2C | 240.2V | - | - |  
14/02/19 | 11:25AM | 1.440kWh | 0.667kWh/kW | 1,215W | 1,200W | 0.556kW/kW | 31.3C | 238.6V | - | - |  
14/02/19 | 11:20AM | 1.340kWh | 0.620kWh/kW | 1,074W | 1,200W | 0.556kW/kW | 31.0C | 237.8V | - | - |  
14/02/19 | 11:15AM | 1.240kWh | 0.574kWh/kW | 983W | 960W | 0.444kW/kW | 30.4C | 237.8V | - | - |  
14/02/19 | 11:10AM | 1.160kWh | 0.537kWh/kW | 839W | 840W | 0.389kW/kW | 30.0C | 237.8V | - | - |  
14/02/19 | 11:05AM | 1.090kWh | 0.505kWh/kW | 696W | 840W | 0.389kW/kW | 29.7C | 236.3V | - | - |  
14/02/19 | 11:00AM | 1.020kWh | 0.472kWh/kW | 368W | 720W | 0.333kW/kW | 29.4C | 234.7V | - | - |  
14/02/19 | 10:55AM | 0.960kWh | 0.444kWh/kW | 760W | 620W | 0.287kW/kW | 29.4C | 234.7V | - | - |  
14/02/19 | 9:55AM | 0.340kWh | 0.157kWh/kW | 661W | 600W | 0.278kW/kW | 26.0C | 232.3V | - | - |  
14/02/19 | 9:50AM | 0.290kWh | 0.134kWh/kW | 626W | 600W | 0.278kW/kW | 25.7C | 233.1V | - | - |  
14/02/19 | 9:45AM | 0.240kWh | 0.111kWh/kW | 552W | 600W | 0.278kW/kW | 24.0C | 233.1V | - | - |  
14/02/19 | 9:40AM | 0.190kWh | 0.088kWh/kW | 510W | 480W | 0.222kW/kW | 23.0C | 232.3V | - | - |  
14/02/19 | 9:35AM | 0.150kWh | 0.069kWh/kW | 467W | 480W | 0.222kW/kW | 21.9C | 230.7V | - | - |  
14/02/19 | 9:30AM | 0.110kWh | 0.051kWh/kW | 414W | 240W | 0.111kW/kW | 21.2C | 231.5V | - | - |  
14/02/19 | 9:25AM | 0.090kWh | 0.042kWh/kW | 113W | 90W | 0.042kW/kW | 19.8C | 230.7V | - | - |  
14/02/19 | 8:25AM | 0.000kWh | 0.000kWh/kW | 42W | 0W | 0.000kW/kW | 15.6C | 229.1V | - | - |  
14/02/19 | 8:20AM | 0.000kWh | 0.000kWh/kW | 34W | 0W | 0.000kW/kW | 14.9C | 229.9V | - | - |  
14/02/19 | 8:15AM | 0.000kWh | 0.000kWh/kW | 30W | 0W | 0.000kW/kW | 14.9C | 229.9V | - | - |  
14/02/19 | 8:10AM | 0.000kWh | 0.000kWh/kW | 23W | 0W | 0.000kW/kW | 14.5C | 229.9V | - | - |  
14/02/19 | 8:05AM | 0.000kWh | 0.000kWh/kW | 0W | 0W | 0.000kW/kW | 13.5C | 230.7V | - | - |  
14/02/19 | 8:00AM | 0.000kWh | 0.000kWh/kW | 0W | 0W | 0.000kW/kW | 13.5C | 230.7V | - | - |  
14/02/19 | 7:55AM | 0.000kWh | 0.000kWh/kW | 0W | - | - | 18.0C | 230.7V | - | -

solmoller, 2019-02-15:
Interesting! Try changing debug level and check the log..

Br Henrik

GPellemans, 2019-02-15:
My info from PVOutput.org of today:

15/02/19 | 10:55AM | 1.220kWh | 0.565kWh/kW | 1,031W | 960W | 0.444kW/kW | 31.2C | 237.8V | - | - |  
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
15/02/19 | 10:50AM | 1.140kWh | 0.528kWh/kW | 991W | 960W | 0.444kW/kW | 30.7C | 236.3V | - | - |  
15/02/19 | 10:45AM | 1.060kWh | 0.491kWh/kW | 968W | 960W | 0.444kW/kW | 30.4C | 237.0V | - | - |  
15/02/19 | 10:40AM | 0.980kWh | 0.454kWh/kW | 945W | 750W | 0.347kW/kW | 30.0C | 237.0V | - | - |  
15/02/19 | 9:40AM | 0.230kWh | 0.106kWh/kW | 522W | 480W | 0.222kW/kW | 24.4C | 231.5V | - | - |  
15/02/19 | 9:35AM | 0.190kWh | 0.088kWh/kW | 483W | 480W | 0.222kW/kW | 23.7C | 230.7V | - | - |  
15/02/19 | 9:30AM | 0.150kWh | 0.069kWh/kW | 443W | 360W | 0.167kW/kW | 22.6C | 230.7V | - | - |  
15/02/19 | 9:25AM | 0.120kWh | 0.056kWh/kW | 388W | 360W | 0.167kW/kW | 21.6C | 229.1V | - | - |  
15/02/19 | 9:20AM | 0.090kWh | 0.042kWh/kW | 329W | 360W | 0.167kW/kW | 20.9C | 230.7V | - | - |  
15/02/19 | 9:15AM | 0.060kWh | 0.028kWh/kW | 295W | 56W | 0.026kW/kW | 20.2C | 230.7V | - | - |  
15/02/19 | 8:10AM | 0.000kWh | 0.000kWh/kW | 21W | 0W | 0.000kW/kW | 15.2C | 227.6V | - | - |  
15/02/19 | 8:05AM | 0.000kWh | 0.000kWh/kW | 0W | 0W | 0.000kW/kW | 14.5C | 228.4V | - | - |  
15/02/19 | 8:00AM | 0.000kWh | 0.000kWh/kW | 0W | 0W | 0.000kW/kW | 14.5C | 227.6V | - | - |  
15/02/19 | 7:55AM | 0.000kWh | 0.000kWh/kW | 0W | 0W | 0.000kW/kW | 14.2C | 229.1V | - | - |  
15/02/19 | 7:50AM | 0.000kWh | 0.000kWh/kW | 0W | - | - | 18.7C | 227.6V | -

I changed debug level to 3 (since 09.58.22)
2019-02-15 09:58:22: Severity 2, Connecting to the serial port
2019-02-15 09:58:22: Severity 2, Asking all inverters to re-register
2019-02-15 09:59:01: Severity 2: got serial B882000A12AB0055 from database
2019-02-15 09:59:01: Severity 2: Try to register known serial number: B882000A12AB0055
2019-02-15 09:59:03: Severity 1:Inverter B882000A12AB0055 acknowledged registration
2019-02-15 09:59:05: Severity 1: Connected to inverter: 1  2000E1.00   TL2000 AS    Ever-Solar      B882000A12AB00553600V610-00009-09
2019-02-15 09:59:15: Severity 2, Asking for any inverters to register
2019-02-15 09:59:17: Severity 3, No response to 'offline query' request - no offline inverters
2019-02-15 09:59:19: Severity 3, B882000A12AB0055 output: 638 W, Total: 12950.9 kWh, Today: 0.42 kWh
2019-02-15 10:01:42: Severity 2, Asking for any inverters to register
2019-02-15 10:01:44: Severity 3, No response to 'offline query' request - no offline inverters
2019-02-15 10:01:46: Severity 3, B882000A12AB0055 output: 693 W, Total: 12951 kWh, Today: 0.44 kWh
2019-02-15 10:04:09: Severity 2, Asking for any inverters to register
2019-02-15 10:04:11: Severity 3, No response to 'offline query' request - no offline inverters
2019-02-15 10:04:13: Severity 3, B882000A12AB0055 output: 714 W, Total: 12951 kWh, Today: 0.47 kWh
2019-02-15 10:06:36: Severity 2, Asking for any inverters to register
2019-02-15 10:06:38: Severity 3, No response to 'offline query' request - no offline inverters
2019-02-15 10:06:40: Severity 3, B882000A12AB0055 output: 710 W, Total: 12951 kWh, Today: 0.5 kWh
2019-02-15 10:09:03: Severity 2, Asking for any inverters to register
2019-02-15 10:09:05: Severity 3, No response to 'offline query' request - no offline inverters
2019-02-15 10:09:07: Severity 3, B882000A12AB0055 output: 746 W, Total: 12951.1 kWh, Today: 0.53 kWh
2019-02-15 10:11:31: Severity 2, Asking for any inverters to register
2019-02-15 10:11:33: Severity 3, No response to 'offline query' request - no offline inverters
2019-02-15 10:11:35: Severity 3, B882000A12AB0055 output: 760 W, Total: 12951.1 kWh, Today: 0.56 kWh
2019-02-15 10:14:00: Severity 2, Asking for any inverters to register
2019-02-15 10:14:02: Severity 3, No response to 'offline query' request - no offline inverters
2019-02-15 10:14:04: Severity 3, B882000A12AB0055 output: 781 W, Total: 12951.1 kWh, Today: 0.59 kWh
2019-02-15 10:16:27: Severity 2, Asking for any inverters to register
2019-02-15 10:16:29: Severity 3, No response to 'offline query' request - no offline inverters
2019-02-15 10:16:31: Severity 3, B882000A12AB0055 output: 799 W, Total: 12951.2 kWh, Today: 0.63 kWh
2019-02-15 10:18:54: Severity 2, Asking for any inverters to register
2019-02-15 10:18:56: Severity 3, No response to 'offline query' request - no offline inverters
2019-02-15 10:18:58: Severity 3, B882000A12AB0055 output: 799 W, Total: 12951.2 kWh, Today: 0.66 kWh
2019-02-15 10:21:21: Severity 2, Asking for any inverters to register
2019-02-15 10:21:23: Severity 3, No response to 'offline query' request - no offline inverters
2019-02-15 10:21:25: Severity 3, B882000A12AB0055 output: 833 W, Total: 12951.2 kWh, Today: 0.69 kWh
2019-02-15 10:23:48: Severity 2, Asking for any inverters to register
2019-02-15 10:23:50: Severity 3, No response to 'offline query' request - no offline inverters
2019-02-15 10:23:52: Severity 3, B882000A12AB0055 output: 843 W, Total: 12951.3 kWh, Today: 0.73 kWh
2019-02-15 10:26:15: Severity 2, Asking for any inverters to register
2019-02-15 10:26:17: Severity 3, No response to 'offline query' request - no offline inverters
2019-02-15 10:26:19: Severity 3, B882000A12AB0055 output: 864 W, Total: 12951.3 kWh, Today: 0.76 kWh
2019-02-15 10:28:42: Severity 2, Asking for any inverters to register
2019-02-15 10:28:44: Severity 3, No response to 'offline query' request - no offline inverters
[... cut ...]
2019-02-15 10:55:49: Severity 3, B882000A12AB0055 uploading to pvoutput.org, response: OK 200: Added Status
2019-02-15 10:58:12: Severity 2, Asking for any inverters to register
2019-02-15 10:58:14: Severity 3, No response to 'offline query' request - no offline inverters
2019-02-15 10:58:16: Severity 3, B882000A12AB0055 output: 1043 W, Total: 12951.8 kWh, Today: 1.27 kWh

On (about) 10.40, 10,45, 10.50 , 10.55 I received a message of an succesfull upload to PVOutput.

Is this information helpful?

Gerard

solmoller, 2019-02-17:
Yes, it does shed some light : you have No uploads between 9:40 and 10:40

But you do recieve data from the invertere in that timeframe:

2019-02-15 10:33:41: Severity 3, B882000A12AB0055 output: 848 W, Total:
12951.4 kWh, Today: 0.87 kWh
(...)
2019-02-15 10:41:02: Severity 3, B882000A12AB0055 output: 945 W, Total:
12951.5 kWh, Today: 0.98 kWh
2019-02-15 10:41:03: Severity 3, B882000A12AB0055 uploading to pvoutput.org,
response: OK 200: Added Status

So either there is some weird bug preventing a post at 10:33, or something
Else in your environment kicks in at  certain times. Are you using the sd
image?

Br Henrik

GPellemans, 2019-02-17:
Hello,

1. Is it an problem of eversolar/rapberry or maybe it's a problem of 
PVOutput.org?
2. Yesterday I changed my API-code: no better results.
3. I changed the intervan from 5 to 10 minutes; no better results.
4. The uploads from today (17 feb 2019):

8.00            10.00
8.10            10.10        12.10
8.20            10.20        12.20                        14.20
8.30            10.30        12.30                        14.30         
                 16.30
                     10.40        12.40 14.40                            
16.40
                                       12.50 14.50                    
         16.50
                                       13.00 15.00                    
         17.00
      15.10                             17.10
                                            17.20

5. I cannot remember using the sd-image. I would say that in 2014 I 
"filled" my raspberry-pi as described in the eversolar-wiki.
6. I don't have any other proram running on the raspberry-pi (as far as 
I know/remember)
7. Is there a log in which you van see whether there is an (attempt) of 
an upload to PVOutput.org. In the eversolar.log you can only see there 
is an answer from PVOutput.

Gr. Gerard

---
Dit e-mailbericht is gecontroleerd op virussen met Avast antivirussoftware.
https://www.avast.com/antivirus

solmoller, 2019-02-17:
Just maybe : try renaming the database..

If it is such an old system, then maybe you have a performance problem. My
system is running on a Raspberry Pi B and has data from 2010 but you never
know

Br Henrik

solmoller, 2019-02-19:
Try changing 
                my $cmd = `curl -m 30 -s -d "d=$pv_date" -d "t=$pv_time" -d "v1=$e_today_wh" -d "v2=$pac" -d "v5=$temp" -d "v6=$volts" -H "X-Pvoutput-Apikey:$pvoutput_api_key" -H "X-Pvoutput-SystemId:$this_pvoutput_system_id" $pvoutput_add_status_url 2>&1 `;                chomp($cmd);
                pmu_log("Severity 3, ".$inverters{$inverter}{"serial"}." uploading to pvoutput.org, response: $cmd");

By adding a log line before the curl line - I'd like to know if it attempts to upload and fails.
Is there data in the database at every 5 minutes? 

BR

Henrik

GPellemans, 2019-02-19:
Hello, as far as I can see your line is the same as the one I have in 
eversolar.pl. The only difference are two quotes: a single quote before 
curl and a single quote before ;chomp . Your line gives a failure in 
starting eversolar.pl.
The line I used (and still use) in eversolar.pl is:

my $cmd = `curl -m 30 -s -d "d=$pv_date" -d "t=$pv_time" -d 
"v1=$e_today_wh" -d "v2=$pac" -d "v5=$temp" -d "v6=$volts" -H 
"X-Pvoutput-Apikey:$pvoutput_api_key" -H 
"X-Pvoutput-SystemId:$this_pvoutput_system_id" $pvoutput_add_status_url 
2>&1 `;                chomp($cmd);
pmu_log("Severity 3, ".$inverters{$inverter}{"serial"}." uploading to 
pvoutput.org, response: $cmd");

You want to add a logline before the curl.  I cannot see where that 
happens? And in which logfile should the information be seen?
I very much appreciate your input in trying solving my "problem". Thank you.
In which database should be every 5 minutes data? If you mean db.sqlite, 
than the answer is yes. If you mean the database at PVOutput.org, than 
the answer is no.

Gr. Gerard

---
Dit e-mailbericht is gecontroleerd op virussen met Avast antivirussoftware.
https://www.avast.com/antivirus

solmoller, 2019-02-20:
Thanks!

If there are data added to the database, then the upload might be the
problem. I suggest you add an extra logging just before the curl upload
line, to see If it is the curl that fails.

 Does a mid-hour restart make it upload?

Br Henrik

GPellemans, 2019-02-20:
Hello, I don't know why, but today there was an upload to pvoutput.org 
every 5 minutes! I cannot remember that I really made a change yesterday 
in eversolar.pl. I only tried to put your suggested codeline in the 
program, but that gave an error when I started eversolar.pl. (Because of 
the quotes, I think). So I went back tot the original line of code.

Only at 14.10 hours today there seemed to be a problem with the readings 
of my eversolar. Also my www. didn't give new readings after 14.10. So 
it wasn't a problem of pvoutput. After reboot of the raspbery pi (at 
about 15.05), everything was alright en the upload continued until sunset.
Tomorrow evening  I let you know whether all uploads went well on Thursday.

You suggest to put an extra logging before the curl-command. I don't 
know anything of perl: what code should I put just before the curl-line?

Thank you,

Gr. Gerard

---
Dit e-mailbericht is gecontroleerd op virussen met Avast antivirussoftware.
https://www.avast.com/antivirus

solmoller, 2019-02-20:
Oh, I thought you knew Perl!

Take 
pmu_log("Severity 3, ".$inverters{$inverter}{"serial"}." uploading to pvoutput.org, response: $cmd");

And modify it to something like

pmu_log("Severity 3, ".$inverters{$inverter}{"serial"}." attempt uploading to pvoutput.org, date: $pv_date  time:$pv_time");
Add that above the curl line, and you'll have a log entry before and after upload

BR

HEnrik

GPellemans, 2019-02-20:
OK. Done. Thank you. And now waiting for sunrise. :)

Gr. Gerard

---
Dit e-mailbericht is gecontroleerd op virussen met Avast antivirussoftware.
https://www.avast.com/antivirus

GPellemans, 2019-02-21:
Hello Henrik,

My logfile gave today every 5 minutes that there was an attempt of an 
upload to pvoutput.org. So that works!
And, the logfile gave every 5 minutes a confirmation of an successful 
upload!

So it looks as though my problem is solved. Because I haven't changed 
anything in the code, except an extra line for the log, it must have 
been a problem on the side of pvoutput.org.

Thank you for your help and advice.

Gr. Gerard

---
Dit e-mailbericht is gecontroleerd op virussen met Avast antivirussoftware.
https://www.avast.com/antivirus

solmoller, 2019-02-21:
No worries, closing

closed 2019-02-21 by solmoller