solmoller/eversolar-monitor #40: Wrong values reported (ZL2000S)

RRJohansen, 2019-03-13:
I followed the excellent guide (thank you very much!) and after some light trouble shooting I got it working. Or so I thought!
I'm seeing wrong values reported in every field (see attached screenshot).

Looking at the inverter directly shows current production of ~70W and Today~2kWh.

Is it possible to directly see the returned string and decode it manually? 
Or how should one proceed?

Thank you in advance. 

  
[image]

solmoller, 2019-03-15:
Interesting! Are you sure you have the right settings for strings? There
are two almost identical protokols...

RRJohansen, 2019-03-16:
You were right. Changing strings to 2 have fixed it. Thank you. 
I was confused by the following text in the guide (https://github.com/solmoller/eversolar-monitor/blob/wiki/Software_installation.md):
"Strings is one for Eversolar TL 2000 and below, it is 2 for TL 3000 and above." ZL2000S is different than TL2000 :-) 

Everything seem to work now. Thank you again.
[image]

solmoller, 2019-03-16:
Thanks for the feedback, I'm changing the ini file and documentation now to reflect the ZL2000S

closed 2019-03-16 by solmoller