solmoller/eversolar-monitor #42: TLC5000 support

stoprocent, 2019-10-19:
Hi Guys 

I have seen there was another issue regarding the same, TLC family inverter (Issue #22) and it was never resolved. I've tried the script today but I'm getting a lot of noise on and I'm not sure what is it.

A lot of ```FF``` or ```FD``` or ```F1```  etc. in quite a regular interval (like 1 sec or less). 
Is it normal behavior before you send a re-register command? I've tried to send commands trough stript or manually tough console the same result. 

The moment I connect the rs485 module to inverter I'm getting constant data on RX (led blinks) but data is rubbish.

I was reading [PMU manual](https://www.zeversolar.com/fileadmin/user_upload/pdf/usermanuals_en/zevercom_user_manual_en.pdf) and it supports both TL family and TLC so I guess communication protocol is same ([as documented here too](https://drive.google.com/file/d/1kzXyG2114c4oZBB9dWruK63_X4J58sIg/view)).

I've tried swapping A and B. Any ideas? 

[image]

stoprocent, 2019-10-19:
I've also tried different methods by dumping and inspecting TCP traffic from the inverter and it's relatively simple to read the data. The inverter is using MQTT (as I investigate it it's Alibaba Cloud platform as it formats the data according to this [documentation](https://www.alibabacloud.com/help/doc-detail/63656.htm)) without TLS so all traffic is unencrypted but the problem is it's sending data every 10-15 min which sucks.

solmoller, 2019-10-20:
That behaviour is not what we see on other inverters. We see dead silence,
unless polling.

Br
Henrik

closed 2021-07-31 by solmoller