solmoller/eversolar-monitor #43: Webserver only showing log window

Reneke43, 2019-10-25:
I've had this problem for quite some time now. I didn't look into it, because the values are pushed to my Domoticz server correctly. But I would like to know what's happening here. I only see the log window and no live output.
[image]

solmoller, 2019-10-25:
Interesting, I've just checked mine, and it works fine. It appears though, that it loads the log first. Try disabling the log display and see if the other parts load..

Reneke43, 2019-10-26:
Uhm, and how/where would I do that? 😳
In the mean time I've tried Safari, Chrome and Firefox on my Mac
and IE, Edge, Chrome and Firefox on Windows; I think it's fair to conclude it's not a browser/OS related issue...

solmoller, 2020-01-09:
Sry for the delay. Go into the code and comment out the log display.

kiekerjan, 2020-01-15:
I got the same issue. In my case it was solved by adding a line in function register_inverter:
```
my $serial_number = unpack( "Z*", $serial_num_response );
$serial_number =~ s/\W//g;
```
(I added the second line, the first is for location reference
Thanks to issue #10

Reneke43, 2020-01-26:
I just saw your comment kiekerjan. I just found out the same after reinstalling the script on a clean Ubuntu server. To prevent a lot of writes to my SSD disk, I disabled the log. The first time I did this on my RPi, I didn't check the web interface. So when I did check it a few weeks later, I didn't connect the dots anymore.
However, I'm no expert at Perl scripting. Is there a way to solve this and get the values for the web view somewhere else? Straight from the variables in stead of the database/log?

solmoller, 2020-01-26:
do you have a db.sqlite file at all? That may be the culprit

Reneke43, 2020-01-26:
Yep, dB.sqlite is present and is being updated, as long as I have set “log = 1” in the ini file. As soon as I set “log = 0”, the webinterface changes.

kiekerjan, 2020-01-27:
@Reneke43 I solved this in the referenced pull request. Unfortunately, It has become quite an involved change, including some database juggling and what not. If you only want that change, you'll still need some perl knowledge. But, I did not have that either when I started this so it shouldn't be too difficult.
I have tested this for about 10 days, and it seems to be working as expected.

commit 1d9fa1ac (2020-02-16): Update eversolar.pl

- Added options to control log cleanup and log output to file
- Perltidy causes a number of whitespace and enters around culry brackets
- Solved bug in send_request which returns FALSE instead of 0 (fixes #48)
- Added loglevel 4 for message contents
- Catch errors when writing to socket or reading from TCP socket, this solves errors caused by remote disconnect
- Added timer delay when searching for TCP socket, when a socket cannot be connected, we wait a bit longer
- Fixed bug where serial number can contain non-ascii characters (fixes #47)
- Use the daily table in the sql [... cut]
closed 2020-02-16 by solmoller