solmoller/eversolar-monitor #45: Configuring for Eversolar Evershine TL5000

cr750, 2020-01-11:
I have an old version of the code (v0.13) working on a PC using eth2ser.
I am attempting to build a Raspberry Pi to take its place.
The Pi install carped on the DBI::sqlite and I had to restart the installetion.
now the code is running in debug=1, the inverter registers, and I'm getting repeated -
 sending packet to inverter...
aa 55 01 00 00 00 10 00 00 01 10
sending packet to inverter...
aa 55 01 00 00 10 11 02 00 01 23
received packet from inverter:
aa 55 00 10 01 00 11 82 32 01 56 02 21 08 ad 01 8c
00 62 00 00 00 55 09 bf 13 89 08 5e ff ff 00 03 56
a8 00 00 5a e3 00 01 00 00 0e a6 ff ff 00 00 00 00
00 00 00 00 ff 00 00 00 0e 00
sending packet to inverter...
aa 55 01 00 00 10 11 02 00 01 23
received packet from inverter:
aa 55 00 10 01 00 11 82 32 01 56 02 22 08 cd 01 88
00 62 00 00 00 56 09 bf 13 89 08 6e ff ff 00 03 56
a9 00 00 5a e3 00 01 00 00 0e a6 ff ff 00 00 00 00
00 00 00 00 ff 00 00 00 0e 2f
sending packet to inverter...
aa 55 01 00 00 00 10 00 00 01 10
sending packet to inverter...
aa 55 01 00 00 10 11 02 00 01 23
received packet from inverter:
aa 55 00 10 01 00 11 82 32 01 57 02 24 08 e6 01 86
00 61 00 00 00 56 09 bf 13 83 08 6a ff ff 00 03 56
a9 00 00 5a e3 00 01 00 00 0e a6 ff ff 00 00 00 00
00 00 00 00 ff 00 00 00 0e 3e
sending packet to inverter...
aa 55 01 00 00 10 11 02 00 01 23
received packet from inverter:
aa 55 00 10 01 00 11 82 32 01 57 02 26 09 3d 01 88
00 5c 00 00 00 55 09 bf 13 80 08 55 ff ff 00 03 56
a9 00 00 5a e3 00 01 00 00 0e a6 ff ff 00 00 00 00
00 00 00 00 ff 00 00 00 0d 7c

solmoller, 2020-01-12:
What is the problem?

solmoller, 2020-01-12:
Are you using the Raspberry image?

cr750, 2020-01-13:
> [
> {
> "@type": "EmailMessage",
> "@type": "ViewAction",
> "target": "https://github.com/solmoller/eversolar-monitor/issues/45?email_source=notifications\u0026email_token=AOH4ASPHFU2RTAELUZ44FCLQ5N5CXA5CNFSM4KFPSDR2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEIXDQGQ#issuecomment-573454362",
> "url": "https://github.com/solmoller/eversolar-monitor/issues/45?email_source=notifications\u0026email_token=AOH4ASPHFU2RTAELUZ44FCLQ5N5CXA5CNFSM4KFPSDR2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEIXDQGQ#issuecomment-573454362",
> "name": "View Issue"
> },
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",
> "url": "https://github.com"
> }
> }
> ]

Hi Henrik,I didn't have logging to DB enabled. A silly mistake on my behalf.I use my own Linux PC web server to run the Perl logger for myself.I'm using a Pi for my father's logger.Regards,PhillSent from my Samsung Mobile on the Telstra Mobile Network

solmoller, 2020-01-16:
Closing, issue is reported in other instances

closed 2020-01-16 by solmoller