solmoller/eversolar-monitor #46: Support for web page scraping

cr750, 2020-01-11:
Hi, I'm installing the monitor on a Pi, and have it working with eth2ser on a TL5000 inverter.
My father has a small eversolar (sorry, don't know the model) that has a web page, /home.cgi, which provides a limited set of data.
I have written some perl to gather data from the page, but I'm having difficulty figuring out how to integrate into eversolar-monitor.

If I provide the screen scraping code, could you integrate it with the main branch? I'm happy to test at any time.
[exampleData.txt](https://github.com/solmoller/eversolar-monitor/files/4049048/exampleData.txt)

[home.php.txt](https://github.com/solmoller/eversolar-monitor/files/4049049/home.php.txt)
[getWebData.pl.txt](https://github.com/solmoller/eversolar-monitor/files/4049050/getWebData.pl.txt)

solmoller, 2020-01-12:
Looks promising, bit why not connect directly to the unknown Eversolar inverter through serial?

cr750, 2020-01-13:
> [
> {
> "@type": "EmailMessage",
> "@type": "ViewAction",
> "target": "https://github.com/solmoller/eversolar-monitor/issues/46?email_source=notifications\u0026email_token=AOH4ASMYLKDL4GG2YROVR73Q5N47ZA5CNFSM4KFQORJKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEIXDPUY#issuecomment-573454291",
> "url": "https://github.com/solmoller/eversolar-monitor/issues/46?email_source=notifications\u0026email_token=AOH4ASMYLKDL4GG2YROVR73Q5N47ZA5CNFSM4KFQORJKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEIXDPUY#issuecomment-573454291",
> "name": "View Issue"
> },
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",
> "url": "https://github.com"
> }
> }
> ]

Hi Henrik,The inverter is in a difficult location for serial cabling, but it does have a Wi-Fi connection from which /home.cgi is available. The Wi-Fi option allows me to configure the Pi, mail it to my 90 year old father, and all he has to do is plug in the Ethernet cable and the power cable.Regards,Phill.Sent from my Samsung Mobile on the Telstra Mobile Network

solmoller, 2021-07-31:
This is getting old, sry

If the inverter is placed within wifi reach, then it should be possible to place a RPI with Wifi near it and connect through serial - or use a serial-wifi adapter, that is supported.

I can't support screen scraping, as I don't have access to an inverter that publishes a web page

BR henrik

solmoller, 2022-01-10:
closed due to inactivity

closed 2022-01-10 by solmoller