solmoller/eversolar-monitor #48: Unending loop when registering inverters

kiekerjan, 2020-01-16:
I encountered an unending loop which brought me quite a few gigabytes of logging ;)
The log shows:
```
Severity 1, Inverter register acknowledgement incorrect for inverter FALSE. Expected 06, received 70
Severity 1, Found serial number: FALSE

```
This seems to be caused by send_request returning FALSE:
```
if (!$sock) {
      return FALSE;
    };
```

This FALSE is then evaluated by register_inverter:

```
$serial_num_response = send_request( 0x00, $CTRL_FUNC_CODES{"REGISTER"}{"OFFLINE_QUERY"} );
    if ($serial_num_response) {
```

Because Perl evaluates the FALSE value to true, register_inverter happily proceeds with trying to register a inverter with serial number FALSE. At which it fails because there is no socket (the reason that send_request returned FALSE to start with).

I think I fixed it by replacing FALSE with 0.
Additionally, I added a break in the aggressive polling loop:
```
#aggresive polling when no inverters connected
		while(keys(%inverters)==0) {
                 register_inverter();
                 if ( !$sock) {
                     last;
                 }
               }
```
Without a socket, register_inverter will never succeed in registering inverters, keeping the %inverters empty and thus the while condition always true.

solmoller, 2020-01-25:
Strange, any idea how this can work for the rest of us?

kiekerjan, 2020-01-25:
Well, if the check for the sock value never triggers, you won't be hit by this. And that should not happen because the inverter connect happens just before it.
I think it has to do with my specific installation. I have powered my rs485-2-Ethernet converter from the rs485 socket on the inverter. However, it seems that during startup (and shutdown) it can happen that not enough power is supplied to the converter, causing it to reboot. This means the socket is gone and would explain why the bug hits my situation.
In the meantime I have been working somewhat more robust code for the send_request sub as well. Once I have tested it enough, I'll share it here.

closed 2020-02-16 by solmoller

solmoller, 2020-02-16:
Thanks for the workover, I was planning to look into tidy and code checking, nice work!

commit 1d9fa1ac (2020-02-16): Update eversolar.pl

- Added options to control log cleanup and log output to file
- Perltidy causes a number of whitespace and enters around culry brackets
- Solved bug in send_request which returns FALSE instead of 0 (fixes #48)
- Added loglevel 4 for message contents
- Catch errors when writing to socket or reading from TCP socket, this solves errors caused by remote disconnect
- Added timer delay when searching for TCP socket, when a socket cannot be connected, we wait a bit longer
- Fixed bug where serial number can contain non-ascii characters (fixes #47)
- Use the daily table in the sql [... cut]