solmoller/eversolar-monitor #50: Connected to inverter, but then communication stops.

pcmalle, 2020-02-19:
I have installed the Perl script on a Windows laptop, and I am using a RS-485 to ethernet converter.

There is connection to one inverter, but then the communication stops.

**The logfile shows:**
2020-02-19 16:24:45: Severity 3, Checking database
2020-02-19 16:24:45: Severity 1, Connecting to the ethernet to serial converter
2020-02-19 16:24:45: Severity 1, Connected to the ethernet to serial converter
2020-02-19 16:24:55: Severity 2, Asking all inverters to re-register
2020-02-19 16:25:14: Severity 1, Found serial number: “Serial number”
2020-02-19 16:25:15: Severity 1, Inverter acknowledged registration
2020-02-19 16:25:16: Severity 1, Connected to inverter: 1  3000E1.00     TL3000                     
“Serial number”
2020-02-19 16:25:16: Severity 2, Asking for any inverters to register

**The last activity in the command window is:**
Sending packet to inverter…
Aa 55 01 00 00 00 10 00 00 01 10
Then nothing more happens.

solmoller, 2020-02-19:
Interesting, can you post a few more packages. 

Looks pretty fine, is it always the same pattern, or intermittent?

pcmalle, 2020-02-20:
I think it is always the same pattern.
Here is some more communication:

[image]

solmoller, 2020-02-22:
That is good data to look at. 

The script should look for other inverters, as some  systems like mine consists more than one inverter, so it looks ok, actually it looks completely as expected, I wonder what happens later.

As part of the Acknowledgement, the script should write the inverter data to the database. Do you have any data in the database ? (I recommend SQLitebrowser for this task)

My guess is that there is a problem writing to the database or reading from is, thus this won't run in line 1026
```
foreach $inverter ( keys(%inverters) ) {
        $response = send_request( $inverter, $CTRL_FUNC_CODES{"READ"}{"QUERY_NORMAL_INFO"} );
        if ($response) {

```
BR Henrik

pcmalle, 2020-02-23:
I I have now managed to get some data shown, but only for a very short time,
then the communication stops again.
There is some data in the database, so the writing to it must be working.
Any ideas how to debug this?

/Matthias

solmoller, 2020-02-23:
You can always post some data here, or fiddle around with some debug code. From your name I guess you're not placed in a sunny part of the world at this time of year, could the inverters simply shut down because of no sun and shitty weather?

pcmalle, 2020-02-23:
You’re absolutely right about the shitty weather, but the inverters status is “Normal”, when I try to connect..

solmoller, 2020-02-23:
Rain? Cant help you there..

Does the data that actually pop up look right?

pcmalle, 2020-02-24:
The data looks fine to me, but apparently I only get 1 line of code,
and then communication stops.

2020-02-24 08:49:34: Severity 3, Checking database
2020-02-24 08:49:34: Severity 1, Connecting to the ethernet to serial converter
2020-02-24 08:49:34: Severity 1, Connected to the ethernet to serial converter
2020-02-24 08:49:44: Severity 2, Asking all inverters to re-register
2020-02-24 08:50:02: Severity 2: got serial B983000A12870063 from database
2020-02-24 08:50:02: Severity 2: Try to register known serial number: B983000A12870063
2020-02-24 08:50:03: Severity 1:Inverter B983000A12870063 acknowledged registration
2020-02-24 08:50:04: Severity 1: Connected to inverter: 1  3000E1.00     TL3000                     B983000A128700633600V610-01011-01
2020-02-24 08:50:14: Severity 2, Asking for any inverters to register
2020-02-24 08:50:15: Severity 1, Found serial number: B983000A12660073
2020-02-24 08:50:16: Severity 1, Inverter acknowledged registration
2020-02-24 08:50:17: Severity 1, Connected to inverter: 1  3000E1.00     TL3000                     B983000A126600733600L610-01005-00
2020-02-24 08:50:18: Severity 3, B983000A12870063 output: 189 W, Total: 19567.9 kWh, Today: 0.08 kWh, 365 days : 19567.9 
2020-02-24 09:20:12: Severity 3, Configfile is: eversolar.ini

2020-02-24 09:20:12: Severity 3, Checking database
2020-02-24 09:20:12: Severity 1, Connecting to the ethernet to serial converter
2020-02-24 09:20:12: Severity 1, Connected to the ethernet to serial converter
2020-02-24 09:20:22: Severity 2, Asking all inverters to re-register
2020-02-24 09:20:40: Severity 2: got serial B983000A12870063 from database
2020-02-24 09:20:40: Severity 2: Try to register known serial number: B983000A12870063
2020-02-24 09:20:41: Severity 1:Inverter B983000A12870063 acknowledged registration
2020-02-24 09:20:42: Severity 1: Connected to inverter: 1  3000E1.00     TL3000                     B983000A128700633600V610-01011-01
2020-02-24 09:20:52: Severity 2, Asking for any inverters to register
2020-02-24 09:20:53: Severity 1, Found serial number: B983000A12660073
2020-02-24 09:20:54: Severity 1, Inverter acknowledged registration
2020-02-24 09:20:55: Severity 1, Connected to inverter: 1  3000E1.00     TL3000                     B983000A126600733600L610-01005-00
2020-02-24 09:20:56: Severity 3, B983000A12870063 output: 245 W, Total: 19568 kWh, Today: 0.19 kWh, 365 days : 19568 

How often is the script supposed to read from the inverter?

pcmalle, 2020-02-24:
Ahh ok. I found the query_inverter_secs = 50 in the ini file.

pcmalle, 2020-02-24:
Could there be a minimum output, before the inverter transmits anything?

And from where comes the ID “Connected to inverter: X”? Do these ID’s have to be different 
from each other, if you have more than one inverter?

solmoller, 2020-02-25:
I don't know of a minimum, my inverters gladly report the very low production these days. The system should be quite ready for more than one inverter, I have two TL 1500's. The ID's are not manually set up.

I see that your system is old, you will have a bad 365 days reading the first year, then that becomes the most interesting parameter at all, as it compares year over year. If you have historical data, it is possible to populate the 'daily' table and fix it.

solmoller, 2020-09-12:
fixed..

closed 2020-09-12 by solmoller