solmoller/eversolar-monitor #51: Eversolar TL1500-20  No response to 'offline query' request - no offline inverters

nicootje55, 2020-10-15:
Hi,

I try to monitor my TL1500-20 with your great Perl program.
The HW-version is v1.1
The SW-version is E1.00.

The RS485 communication is working fine but after receiving the serial number of my inverter i get the message No response to 'offline query' request - no offline inverters.

The log:
2020-10-15 19:01:54: Severity 1, Connecting to the serial port
SolarWebServer: You can connect to your server at http://localhost:3837/
2020-10-15 19:01:54: Severity 2, Asking all inverters to re-register
sending packet to inverter...
aa 55 01 00 00 00 10 04 00 01 14
sending packet to inverter...
aa 55 01 00 00 00 10 04 00 01 14
sending packet to inverter...
aa 55 01 00 00 00 10 04 00 01 14
sending packet to inverter...
aa 55 01 00 00 00 10 04 00 01 14
sending packet to inverter...
aa 55 01 00 00 00 10 04 00 01 14
sending packet to inverter...
aa 55 01 00 00 00 10 04 00 01 14
sending packet to inverter...
aa 55 01 00 00 00 10 04 00 01 14
sending packet to inverter...
aa 55 01 00 00 00 10 04 00 01 14
2020-10-15 19:02:12: Severity 1, done registering known inverters
sending packet to inverter...
aa 55 01 00 00 00 10 00 00 01 10
received packet from inverter:
aa 51 00 00 01 00 10 80 12 58 48 31 31 30 30 32 30
31 31 31 31 30 31 30 32 00 01 00 fa
2020-10-15 19:02:13: Severity 3, fallback to no response
2020-10-15 19:02:13: Severity 3, No response to 'offline query' request - no offline inverters
sending packet to inverter...
aa 55 01 00 00 00 10 00 00 01 10
2020-10-15 19:02:15: Severity 3, fallback to no response
2020-10-15 19:02:15: Severity 3, No response to 'offline query' request - no offline inverters 

The strange thing is that the inverter responds wint AA 51 and not with AA 55.
I have read the inverter protocol and tried a lot of control and function codes but cannot fix this problem. 
The inverter does not respond on all those command-packets.

Please help.

Nico Kooij

solmoller, 2020-10-15:
Maybe the protol is change. How old is your inverter?

Try hacking it and change the ack from 55 to 51

Br Henrik

nicootje55, 2020-10-15:
Thank you Henrik.

 

The inverter is about six years old.

 

I will try this hack tomorrow and let you know.

 

Nico Kooij

nicootje55, 2020-10-16:
Hello Henrik,

I tried this hack today.

With this hack i receive nothing anymore.

See the log below:

2020-10-16 17:54:00: Severity 1, Connecting to the serial port
SolarWebServer: You can connect to your server at http://localhost:3837/
2020-10-16 17:54:00: Severity 2, Asking all inverters to re-register
sending packet to inverter...
aa 51 01 00 00 00 10 04 00 01 10
sending packet to inverter...
aa 51 01 00 00 00 10 04 00 01 10
sending packet to inverter...
aa 51 01 00 00 00 10 04 00 01 10
sending packet to inverter...
aa 51 01 00 00 00 10 04 00 01 10
sending packet to inverter...
aa 51 01 00 00 00 10 04 00 01 10
sending packet to inverter...
aa 51 01 00 00 00 10 04 00 01 10
sending packet to inverter...
aa 51 01 00 00 00 10 04 00 01 10
sending packet to inverter...
aa 51 01 00 00 00 10 04 00 01 10
2020-10-16 17:54:18: Severity 1, done registering known inverters
sending packet to inverter...
aa 51 01 00 00 00 10 00 00 01 0c
2020-10-16 17:54:20: Severity 3, fallback to no response
2020-10-16 17:54:20: Severity 3, No response to 'offline query' request - no offline inverters
sending packet to inverter...
aa 51 01 00 00 00 10 00 00 01 0c
2020-10-16 17:54:22: Severity 3, fallback to no response
2020-10-16 17:54:22: Severity 3, No response to 'offline query' request - no offline inverters

Do you have more suggestions?

Thanks in advance,

Nico Kooij

solmoller, 2020-10-16:
Try a split hack: send aa 55 and interpret aa 51 as a good reply...

Do you know anyone else with eversolar inverter?

Br Henrik

nicootje55, 2020-10-16:
Hello Henrik,

I tried your suggestion and we are getting closer but no sulution yet.

See the log below:

2020-10-16 18:19:25: Severity 1, Connecting to the serial port
SolarWebServer: You can connect to your server at http://localhost:3837/
2020-10-16 18:19:26: Severity 2, Asking all inverters to re-register
sending packet to inverter...
aa 55 01 00 00 00 10 04 00 01 14
sending packet to inverter...
aa 55 01 00 00 00 10 04 00 01 14
sending packet to inverter...
aa 55 01 00 00 00 10 04 00 01 14
sending packet to inverter...
aa 55 01 00 00 00 10 04 00 01 14
sending packet to inverter...
aa 55 01 00 00 00 10 04 00 01 14
sending packet to inverter...
aa 55 01 00 00 00 10 04 00 01 14
sending packet to inverter...
aa 55 01 00 00 00 10 04 00 01 14
sending packet to inverter...
aa 55 01 00 00 00 10 04 00 01 14
2020-10-16 18:19:44: Severity 1, done registering known inverters
sending packet to inverter...
aa 55 01 00 00 00 10 00 00 01 10
received packet from inverter:
aa 51 00 00 01 00 10 80 12 58 48 31 31 30 30 32 30
31 31 31 31 30 31 30 32 00 01 00 fa
2020-10-16 18:19:45: Start modified repons routine: dont ceck checksum (line 437 in perl script)
2020-10-16 18:19:45: Severity 3, Unpacked received serial number: XH11002011110102 .
2020-10-16 18:19:45: Severity 1, Cleaned received serial number: XH11002011110102 .
sending packet to inverter...
aa 55 01 00 00 00 10 01 13 58 48 31 31 30 30 32 30
31 31 31 31 30 31 30 32 00 01 10 04 80
2020-10-16 18:19:47: Severity 3, fallback to no response
2020-10-16 18:19:47: Severity 2, No response to 'send register address' request for inverter XH11002011110102
sending packet to inverter...
aa 55 01 00 00 00 10 00 00 01 10
2020-10-16 18:19:49: Severity 3, fallback to no response
2020-10-16 18:19:49: Severity 3, No response to 'offline query' request - no offline inverters

Do you have more suggestions?

Thanks in advance,

Nico Kooij

solmoller, 2020-10-16:
Are the seriel numbers correctly read?

Br Henrik

nicootje55, 2020-10-16:
Hello Henrik,

I really appreciate that you are trying to help me.

My serial number is XH11002011110102. That is: 58 48 31 31 30 30 32 30 31 31 31 31 30 31 30 32.
So that is correct.

After that i see a protocol version of 0001 and a checksum of 00 FA.
Or is there something wrong with that?

I do not understand why the inverter is not responsing to this command.

Do you?

Nico Kooij

solmoller, 2020-10-16:
Interesting! I am traveling, so I don't have any documentation here. You
are moving in the right direction, so try and experiment a little

Br Henrik

nicootje55, 2020-10-17:
Hallo Henrik,

I tried a lot in sending adress assign (10 01) after receiving the serial number but did not get a respons from my inverter.

As i told before my inverter returns the next serial-number respons:
aa 51   00 00   01 00   10 80   12   58 48 31 31 30 30 32 30 31 31 31 31 30 31 30 32   00 01   00 fa

It is strange that it returns AA 51 and not AA 55.  That is not like protocol version 2.0.2 (15.9.21).

My serial number is XH11002011110102
Thats the part 58 till 32, 16 positions and is correct.

It is also strange that it returns 00 01 after the serial number. That is not like protocol version 1.1.
Is this telling me that it uses protocol version 1.0? Does that exist and is there a protocol explanation for that?

It is also strange that the monitor sees the checksum as invalid. 

by default the monitor sends the next record after the above respons:
aa 55   01 00   00 00   10 01   13   58 48 31 31 30 30 32 30 31 31 31 31 30 31 30 32   00 01   10   04 80

I tried the following without success:
- changed aa 55 in aa 51	(at the beginning)
- changed 00 01 in 01		(at the end before the checksum)
- changed dest address in 00 01

Do you have a suggestion what i can try next?
Are there more protocol versions in PDF out there?

Thanks in advance.

Nico Kooij

nicootje55, 2020-10-17:
Sorrie Henrik, i did not look well to my serial number and thus lied to you.

My real serial number is XH15006011550102 
The monitor sees serial  XH11002011110102  That differs 4 for some characters. 

That is very strange beheviour.
I will test it with the real serial but do you understand this behaviour?

Thanks,

Nico Kooij

solmoller, 2020-10-17:
If you look at response 55 and 51 are mixed
If you look at serial, 1's And 5's respectively 2's and 6's are affected

Could be a faulty circuit board with a bad third bit

Henrik

nicootje55, 2020-10-17:
I am sure you are right Henrik, i have a faulty circuit board..

It worked today for a very short time after i unplugged the RS485 to USB interface and replugged it again.

I will buy a new board.

Thanks a lot for your comments. I will close this issue.

Nico Kooij

solmoller, 2020-10-17:
The fault could be on the inverter...

Br Henrik

nicootje55, 2020-10-17:
Hello Henrik,

Do you really think so?

When i unplug my USB to RS485converter and replug it again i get the next behaviour. It starts working well but after a while it looses connection. At that moment i see again the 0x51 and not the 0x55 in the receive.

When i restart the monitor it keeps giving 0x51 and does not work until i restart the USB to RS485 converter and the monitor.

2020-10-17 17:51:08: Severity 1, Connecting to the serial port
SolarWebServer: You can connect to your server at http://localhost:3837/
2020-10-17 17:51:08: Severity 2, Asking all inverters to re-register
sending packet to inverter...
aa 55 01 00 00 00 10 04 00 01 14
2020-10-17 17:51:10: Severity 2, got serial XH15006011550102 from database
2020-10-17 17:51:10: Severity 2, Try to register known serial number: XH15006011550102
sending packet to inverter...
aa 55 01 00 00 00 10 01 11 58 48 31 35 30 30 36 30
31 31 35 35 30 31 30 32 10 04 8d
received packet from inverter:
aa 55 00 10 01 00 10 81 01 06 01 a8
2020-10-17 17:51:11: Severity 1, Inverter XH15006011550102 acknowledged registration
sending packet to inverter...
aa 55 01 00 00 10 11 03 00 01 24
received packet from inverter:
aa 55 00 10 01 00 11 83 4d 31 20 20 31 35 30 30 45
31 2e 30 30 20 20 20 54 4c 31 35 30 30 2d 32 30 20
20 20 20 45 76 65 72 2d 53 6f 6c 61 72 20 20 20 20
20 20 58 48 31 35 30 30 36 30 31 31 35 35 30 31 30
32 33 36 30 30 56 36 31 30 2d 30 31 30 32 33 2d 30
36 12 48
2020-10-17 17:51:12: Severity 1, Connected to inverter: 1  1500E1.00   TL1500-20    Ever-Solar      XH150060115501023600V610-01023-06
2020-10-17 17:51:22: Severity 1, done registering known inverters
2020-10-17 17:51:22: Severity 2, Asking for any inverters to register
sending packet to inverter...
aa 55 01 00 00 00 10 00 00 01 10
2020-10-17 17:51:24: Severity 3, fallback to no response
2020-10-17 17:51:24: Severity 3, No response to 'offline query' request - no offline inverters
sending packet to inverter...
aa 55 01 00 00 10 11 02 00 01 23
received packet from inverter:
aa 55 00 10 01 00 11 82 2c 00 7d 00 2e 04 43 00 01
00 01 09 12 13 86 00 05 ff ff 00 01 61 74 00 00 50
f3 00 01 00 00 00 00 ff ff 00 00 00 00 00 00 ff 00
00 00 0a 91
2020-10-17 17:51:25: Severity 3, select min(e_total) as mins from daily where timestamp  >=   date('now', '-1 year') and serial_number = XH15006011550102
2020-10-17 17:51:25: Severity 3, etot = 9048.3
2020-10-17 17:51:25: Severity 3, Operation done successfully
2020-10-17 17:51:25: Severity 3, XH15006011550102 output: 5 W, Total: 9048.3 kWh, Today: 0.46 kWh, 365 days : 0
2020-10-17 17:51:25: Severity 3, uploading to pv-log.com, data: m[mi++]="17.10.20 17:51:00|5;10;460;109;12"

2020-10-17 17:52:15: Severity 2, Asking for any inverters to register
sending packet to inverter...
aa 55 01 00 00 00 10 00 00 01 10
2020-10-17 17:52:17: Severity 3, fallback to no response
2020-10-17 17:52:17: Severity 3, No response to 'offline query' request - no offline inverters
sending packet to inverter...
aa 55 01 00 00 10 11 02 00 01 23
received packet from inverter:
aa 51 00 10 01 00 11 82 28 00 79 00 2a 00 89 00 01
00 02 09 12 13 82 00 00 fb fb 00 01 61 70 00 00 50
f3 00 01 00 00 00 00 fb fb 00 00 00 00 00 00 fb 00
00 00 0a db
2020-10-17 17:52:18: Severity 3, fallback to no response
2020-10-17 17:52:18: Severity 2, XH15006011550102 lost contact with inverter (1 time(s))
2020-10-17 17:53:08: Severity 2, Asking for any inverters to register

What do you suggest now?

Thanks in advance,

Nico Kooij

solmoller, 2020-10-17:
Adapters should be cheap, I'd try another one

Br Henrik

nicootje55, 2020-10-20:
Hello Henrik,

I tried another RS485 adapter and the communication is fine now. I tested this with Python.

But there is one new problem: this adapter is not working with Perl.

When eversolar.pl opens the serial port it gives the next message:
2020-10-20 12:41:37: Severity 1, Connecting to the serial port
SolarWebServer: You can connect to your server at http://localhost:3837/
invalid COMMPROP block length= 66 at D:\eversolar\eversolar.pl line 542.
Can't open : No such file or directory at D:\eversolar\eversolar.pl line 542.

The RS485 adapter is based on CP2102. I tested an other USB2serial adapter based on CP2102 and there i got the same error message. So i think it has to do with the CP2102 chip.

I hope you have an idea how to solve this. Otherwise i will make my own Python program. 

Thanks in advance,

Nico Kooij

closed 2021-05-06 by solmoller