solmoller/eversolar-monitor #56: Overnight stops updating

nagydavid, 2021-12-21:
Hi @solmoller,

I just want to explain a bit better my question, which I mentioned in the PR.
Specs.:
Hardware.:
- Zeverlution 2000S
- Raspberry Pi Zero W with a USB to RS485/RS422.
Software: 
- Raspberry Pi OS bullseye
- Perl v5.32.1
- Manual ES-monitor Installation

When there is production(sun) everything works. Getting data, `sub main_loop()` operates, etc.
But when production seize (night), then the loop seems to freeze.
Also when I restart the script, during offline ours then I get the following:

```
2021-12-21 19:47:04: Severity 1, Configfile is: eversolar.ini
Update old database version, or print 2 fail messages (part 1) : 
DBD::SQLite::db do failed: duplicate column name: vpv2 at eversolar.pl line 874.
DBD::SQLite::db do failed: duplicate column name: ipv2 at eversolar.pl line 875.
Done updating old database version.  
2021-12-21 19:47:05: Severity 3, Checking database
Update old daily database version, or print 2 fail messages (part 2) : 
DBD::SQLite::db do failed: duplicate column name: pmax_today at eversolar.pl line 918.
DBD::SQLite::db do failed: duplicate column name: pmax_time at eversolar.pl line 919.
Done updating old daily database version.  
2021-12-21 19:47:05: Severity 1, Connecting to the serial port
2021-12-21 19:47:05: Severity 2, Asking all inverters to re-register
sending packet to inverter... 
aa 55 01 00 00 00 10 04 00 01 14 
sending packet to inverter... 
aa 55 01 00 00 00 10 04 00 01 14 
sending packet to inverter... 
aa 55 01 00 00 00 10 04 00 01 14 
sending packet to inverter... 
aa 55 01 00 00 00 10 04 00 01 14 
sending packet to inverter... 
aa 55 01 00 00 00 10 04 00 01 14 
sending packet to inverter... 
aa 55 01 00 00 00 10 04 00 01 14 
sending packet to inverter... 
aa 55 01 00 00 00 10 04 00 01 14 
sending packet to inverter... 
aa 55 01 00 00 00 10 04 00 01 14 
2021-12-21 19:47:23: Severity 1, done registering known inverters
sending packet to inverter... 
aa 55 01 00 00 00 10 00 00 01 10 
2021-12-21 19:47:25: Severity 3, fallback to no response
2021-12-21 19:47:25: Severity 3, No response to 'offline query' request - no offline inverters
sending packet to inverter... 
aa 55 01 00 00 00 10 00 00 01 10 
2021-12-21 19:47:27: Severity 3, fallback to no response
2021-12-21 19:47:27: Severity 3, No response to 'offline query' request - no offline inverters
...
...
last four line repeats itself, with increasing intervals of (10-30-40-50....etc.)

```

My config looks the following.:
```
[options]
query_inverter_secs = 50                # how often to request data from the inverter and write to log file
debug = 4                               # Debug level , 0 = none, 1 is high priority, 3 is everything, 4 also output packets sent to and received f>
log_file = "/var/log/eversolar"         # make sure the user executing the script can write to this file
output_to_log = 1                       # 1 = output to the above logfile, 0 do not
clean_log = 1                           # Cleanup log file every month, 0 = disable, 1 = enabled (default)
communication_method = "serial" # "eth2ser" or "serial". defines wether you are using an ethernet to serial converter to connect to the inverter, o>
strings = 2      # number of PV panel strings (panels wired together) connected to your inverter. 1 or 2 supported.

[serial]                                # communication_method must be set to "serial" above for serial comms
port = "/dev/ttyUSB0"
```
Apart from my newly proposed MQTT feature, everything is disabled.

I just would like to ask to take a peak if I have made some error

solmoller, 2021-12-21:
That all looks very fine, both log and config

I suggest you simulate 'night' 

There are many feasible ways of doing this, some even faster than waiting for an eclipse. Some are simpler than throwing a blanket on the panels. 
At night the inverter is powerless, but I think that unplugging the serial cable while the system is not communicating may have the same effect to your software.

It's the darkest day of the year here in Denmark, we have plenty of 'night'

nagydavid, 2021-12-21:
perfect, i will try it out tomorrow :)

nagydavid, 2022-01-08:
I'm closing, I think it was related to the influx setting which halted the update, when I turned it off, and only the mqtt is enabled, everything worked as intended

closed 2022-01-08 by nagydavid