solmoller/eversolar-monitor #58: pi running eversolar monitor loses connection

LeighAS, 2022-06-12:
https://pvoutput.org/list.jsp?p=0&id=67748&sid=60211&v=0&o=date&d=desc

I have been running this script on an early pi for years now. very recently I have added more panels and every few days or so my pi needs a reboot to re-establish connection to the inverter.

I wondered whether its a rs485 issue as I don't use any terminating resistors but the wire is only 1metre long and its is a cat5 cable with small jumper wires from a cat5 socket into an rs485 to usb dongle.

eventually id love to get this working flawlessly with 3 (or more) arrays.
I have an 4.4kw dual string inverter and a 3.6kw single string inverter and an option to buy some smaller 2kw eversol inverters too.

I would also like to have my car (2011 muxsan leaf) alter its charge profile on the fly to soak up the total production by the car communicating with pi to use as much solar as possible without dipping into importing from the grid.

thanks for all your hard work here.

solmoller, 2022-06-12:
Hi Leigh,

Did you use the SD card image? I ran that image for nine years on a Rpi model B almost flawlessly. I've started over on an RPi4, but the image did not transfer, so I started over. Maybe your database is too large for the old HW?

As for car integration, I'd look into getting the newer code to deliver to a Home Assistand, and make any automations to the Leaf from there.

Happy nerding!

Henrik

LeighAS, 2022-06-12:
might be the database on a model A pi just becoming too big after a few days but then I wonder whether if that's true a bigger memory of a more recent pi would eventually fall over too? just take longer before it does?

Will defo grab a latest code to do the Hass stuff when doing the leaf work its something I've been meaning to get into. having some chargers for small mower batteries (i use for my ebikes) come on at lower solar outputs would be easy with the mqtt stuff. currently using tuya but I have some tasmotized sonoff switches which I can use so this job too.

solmoller, 2022-06-12:
Do you reset the entire database? Can't be true. The database should hold all your production data.

LeighAS, 2022-06-12:
I just turn it off and on again and it works for a few more days.

LeighAS, 2022-06-13:
altered my ini last night to increase logging to level 4 from 1 and at the same time I changed the strings number to 2 as the tl4600 I have actually has a second dc input than its predecessor which I'm now using with the second array. Unfortunately I couldnt test till this morning and then when i checked the watts and the ac frequency were all wrong so I've reverted that change to the ini and kept the logging level at 4.

once ive got a few days logging and it falls over again can i send you the logs for you to tell me what i might be doing wrong? + setup ini, redacted, thanks.

LeighAS, 2022-06-15:
tried dragging and dropping files but doesn't appear to work

LeighAS, 2022-07-14:
pi@raspberrypi:~/eversolar-monitor $ sudo perl eversolar.pl
mqtt_inverter_model expects an argument at eversolar.ini line 51
2022-07-14 10:57:53: Severity 1, Configfile is: eversolar.ini
Update old database version, or print 2 fail messages (part 1) :
Done updating old database version.
2022-07-14 10:57:55: Severity 3, Checking database
2022-07-14 10:57:55: Severity 1, Updating database with daily table, making data export and stats easy
SolarWebServer: You can connect to your server at http://localhost:3837/
2022-07-14 10:57:56: Severity 1, Connecting to the serial port
**Can't open : No such file or directory at eversolar.pl line 561.**
pi@raspberrypi:~/eversolar-monitor $ mqtt_inverter_model expects an argument at eversolar.ini line 51
2022-07-14 10:57:53: Severity 1, Configfile is: eversolar.ini
Update old database version, or print 2 fail messages (part 1) :
Done updating old database version.
2022-07-14 10:57:55: Severity 3, Checking database
2022-07-14 10:57:55: Severity 1, Updating database with daily table, making data export and stats easy
SolarWebServer: You can connect to your server at http://localhost:3837/
2022-07-14 10:57:56: Severity 1, Connecting to the serial port
Can't open : No such file or directory at eversolar.pl line 561.

This is the output now at putty's terminal through ssh when I run the perl script. my suspicion is that the serial to usb device may not be working but its was initially. I have killed the process a few times to attempt to solve this issue but I wonder whether I have aggravated the issue by maybe corrupting the sql lite db.

still having issues on both setups it appears and haven't managed to narrow down the source of the instability.

LeighAS, 2022-07-14:
affirmed my suspicion of the hardware issues by switching the usb to rs485 adapters between my two working pi (one old software, one new software). The newest software got past the above issue at line 561 now and as started seeing the inverter again. Strangely the old software pi seem to be okay too. I am starting to think this might be a really silly issue like insufficiently capable 5v psu or temperamental usb to rs485 chips.

would really like to get past these instability issues so i can work on the rest of my system.

LeighAS, 2022-07-14:
https://pvoutput.org/intraday.jsp?id=67748&sid=92094

This is my newly aggregated system, I have both inverters separately pushing the data via two different pi. One has the old bug of being an hour forward but this doesn't seem to be there in the newer software. I don't mind, today was good in that I now have a simple way of seeing the daily production in one place and if I find another eversol inverter, a way of adding more systems in the future. The aggregation and multiple systems feature cost a donation or minimum $15australian but that is pretty good value.

I could update the old firmware now but equally i will soon move the inverter next to the other one and run them both off a single pi with chained rs485 bus. This will have to wait as corona has got me grabbing low hanging software fruit jobs.

wotid, 2022-07-16:
Leigh, 

I have my Eversol on PVOutput running this software too. https://pvoutput.org/intraday.jsp?id=2154&sid=12337&dt=20220716 I use MySQL rather than SQlite. (I also turned off the frontend).

wotid, 2022-07-19:
I liked the front end too, but run one that's closer (sort of) to PVOutput.
[image]
[image]

LeighAS, 2022-07-25:
Pi seems to be stable now running two inverters on newest firmware. as this is the case i will close this issue and open a new one specifically to deal with the totals not being output correctly for dual inverters

closed 2022-07-25 by LeighAS