solmoller/eversolar-monitor #59: Full integration with the new Home Assistant 'Energy' tab

solmoller, 2022-06-15:
The integration to HA is good, but we should look into a full and automated integration

sorentorp, 2022-06-20:
Yay - great idea!

LeighAS, 2022-07-22:
Hey peeps, spent some time tonight getting Home assistant setup on my rp3. Installed mosquitto mqtt broker addon and configured it, then altered my ini file to match on the eversolar rpi.
[image]
[image]

I don't seem to be able to hear anything the eversolar monitor is publishing from the mqtt broker. what am I missing?

LeighAS, 2022-07-22:
sudo apt install -y mosquitto mosquitto-clients

just read this in the comments of the perl script. definitely missed that. also now the sun is down i think i'm missing the light too.

LeighAS, 2022-07-22:
[image]
managed to send a message from the clients cmd line and received it in homeassistant so its looking promising now for tomorrow when the sun comes up, hoping auto discovery works

LeighAS, 2022-07-23:
has been working since this morning except I think there may still be issues regarding running 2 inverters through one pi.

[image]
so this is the webpage vs home assistant.
the daily total so far appears correct but the combined instantaneous power is incorrect, this would be a useful metric to have so as to use for storing power.

My shelly EM arrived today and that can give me the import/export values for the whole house combined so I will be most likely using those values for turning on loads i.e. batteries or turning up a hybrid inverter if the sun is not out and I am importing.

Thankfully pvoutput is correct as this is summed/aggregated on the website after being sent up via two separate api calls.

I am really impressed with all the improvements. Thanks for everyone's hard work here.

nagydavid, 2022-07-25:
I am going on holiday, so I will have time to tinker, can you please guide me, what additional integration you would like with home assistant? @solmoller

LeighAS, 2022-07-26:
[image]
managed to create a binary sensor in home assist today which will detect whether the last update time was over 45 secs ago. This should be a quick way of me finding out at a glance whether the usb to rs485 stick as been disturbed and is no longer collecting infos. I think this has been the issue since the start and today was the first time this had broken since sorting out the inverters both onto a single rpi on the same rs485 bus. The reason i caught it was that i had shifted a 12v battery for charging and that was under the rpi which was then an obvious timestamp for me to show why id stopped getting solar data. Anyway i should be able to automate some trigger to notify me when this sensor changes outside of night time hours.

nagydavid, 2022-07-27:
I have modified the mqtt messages in order to be inline with the timestamp requirements of home assistant. I will update my code somewhere in this week.[image]

LeighAS, 2022-07-27:
nice, that will break my new pv connected widget but its a better way to have it.
As an aside The setting page of my binary sensor gives a nice depiction of the uptime of the inverters in green for on and red for off.

@nagydavid Any chance you could have a look at the multiple inverters bug for totals? I have got a 3rd eversolar inverter on the way so will be able to test with all 3 eventually

nagydavid, 2022-07-27:
Well I only have one inverter, but if you manage to print out the inverters value into a json I will be able to look into it. @solmoller Any suggestions to do that?

solmoller, 2022-08-04:
I'll look at it when other projects permit :-)

LeighAS, 2022-12-06:
just found a work around for now to my energy dash board issue with the mqtt from the eversolar monitor not being formatted exactly as home assistant expects.

[image]
the combined energy which is the summation of the 2 inverters by the perl code sent via mqtt doesnt have a last_reset.

instead i've found that i can use pvoutput.org integration and that gives me accurate summation of the solar albeit having to round trip the data to and from the pvoutput servers. Its also added together by pvoutput.org.

also in re investigating this i've realised that you can add multiple solar sources and home assistant energy bar graphs end up stacked, so using sensor.pv_total_energy_production and sensor.pv_total_energy_production_2 as inputs i now get accurate totals and nicely stacked graphs.
[image]

Timdebruijn, 2024-01-18:
> I have modified the mqtt messages in order to be inline with the timestamp requirements of home assistant. I will update my code somewhere in this week.[image]

Hi! What did you change to get this working? Looks like it's still broken(fresh install)

nagydavid, 2024-01-18:
i need to look in to that it was long time ago

Timdebruijn, 2024-01-18:
I Think it had something to do with the formatting of the timestamp, currently it's something like:
2024-01-18 17:25:16
But maybe it has to be correctly formatted in ISO 8601:
2024-01-18T17:25:16Z

I've already tried that, but not with any succes.. 
on line 831:
`sub get_timestamp {
    ( $sec, $min, $hour, $mday, $mon, $year, $wday, $yday, $isdst ) = localtime(time);
    return sprintf( "%04d-%02d-%02dT%02d:%02d:%02dZ", $year + 1900, $mon + 1, $mday, $hour, $min, $sec );
}`

Timdebruijn, 2024-01-19:
@nagydavid
I've tried many things like:
```
use POSIX qw(strftime);

sub get_timestamp {
    my $timestamp = strftime("%Y-%m-%dT%H:%M:%S%z", localtime(time));
    return $timestamp;
}
```
Or
`    use POSIX qw(strftime);
    my $timestamp = strftime("%Y-%m-%dT%H:%M:%SZ", gmtime);`

But none of this gets the status in Home assisant other than "Unknown"

Do you have any idea in what direction I should look?

nagydavid, 2024-01-19:
I just checked my code, and it runs as intended
[image]

Timdebruijn, 2024-01-19:
@nagydavid Thank you so much!!

I think you still need to remove all the extra + symbols..

nagydavid, 2024-01-22:
can you be more specific about the symbols?

Timdebruijn, 2024-01-22:
@nagydavid If you look at https://github.com/solmoller/eversolar-monitor/blob/81898e6f225c4ce7edd26aeb560b75f49501dcd9/eversolar.pl#L1265

You can see that you've added the + symbols to the code.

nagydavid, 2024-01-27:
@Timdebruijn it is fixed.

solmoller, 2024-02-11:
I just looked at this today, and my data is present i HA, but not on the energy Dashboard.

nagydavid, 2024-02-11:
Funny, I added long time ago and it works beautifully.
[image]

nagydavid, 2024-02-11:
@solmoller If you r using the mqtt integration then PV Total Energy Production adheres to the requirements regarding device_class and state_class. So if you add that entity to ur dashboard it should work.

tobiasnorbo, 2024-05-02:
Hi Guys, @nagydavid and @Timdebruijn 

Could some of you make a pastebin of the changes to the ini file, and maybe discribe the setup of mosquito MQTT broker, so it is easier for people like me to follow in your foot steps and start improving on what has already been created?

Would be highly appreciated.

Thanks :)

solmoller, 2024-05-03:
GitHub can compare the file versions for you :-)

Br Henrik

tobiasnorbo, 2024-05-03:
> sudo apt install -y mosquitto mosquitto-clients
> 
> just read this in the comments of the perl script. definitely missed that. also now the sun is down i think i'm missing the light too.

I am not able to install this on the RPI2, that I am running the script on. Also I have to write: sudo apt-get install -y mosquitto mosquitto-clients

[image]

nagydavid, 2024-05-03:
@tobiasnorbo it  is written in your error message: update or do the fix missing. By the way if you have some linux related problem, try chatgpt it is quite good pointing you the answer.

tobiasnorbo, 2024-05-03:
Sorry, I should have written that as well. This is the messages I get when trying to update and fix. Doesn't seem to work. 

If you have a tip please let me know. otherwise I will get chatgpt to help. 
[image]

tobiasnorbo, 2024-05-03:
Just an update for anybody getting to this point as well. If you consider using the software in HA, then you cannot use the img file. You will need an up to date OS. 

Therefore I will suggest to run it on a PC, that are also running your HA instance or a newer RPI, since you need to go through the full process anyways, if you wish to go for HA integration.

If you do this, please consider making a img file for a newer RPI, which can be used for future users of the software. I'm not sure people will go for a RPI2 in the future, knowing this, so I will not make an img file for my RPI2 setup, when it is up and running.