solmoller/eversolar-monitor #60: when trying to run latest build from source i get the following error

LeighAS, 2022-07-06:
pi@raspberrypi:~/eversolar-monitor $ sudo perl eversolar.pl
mqtt_inverter_model expects an argument at eversolar.ini line 51
2022-07-06 14:14:36: Severity 1, Configfile is: eversolar.ini
install_driver(SQLite) failed: Can't locate DBD/SQLite.pm in @INC (you may need toueabihf/perl/5.32.1 /usr/local/share/perl/5.32.1 /usr/lib/arm-linux-gnueabihf/perlabihf/perl/5.32 /usr/share/perl/5.32 /usr/local/lib/site_perl) at (eval 14) line 3
Perhaps the DBD::SQLite perl module hasn't been fully installed,
or perhaps the capitalisation of 'SQLite' isn't right.
Available drivers: DBM, ExampleP, File, Gofer, Mem, Proxy, Sponge.
 at eversolar.pl line 275.

ive tried to get the DBD-SQLite from cpanm
both using just
sudo cpanm DBD::SQLite
and
when that hangs at teh building and testing stage and earlier version 1.66 using
sudo cpanm DBD::SQLite~">=1.6600,<=1.6700"

both seem to take forever

running again with the verbose flag on v1.70 and -n for no tests. Seems to take forever at each build step. perhaps thats because this is the earliest raspberry pi?

wonder is there is a way to build this on an emulated pi (high speed) and just copy the binaries over. I'm not so clued up with linux but this is taking the pi ages.

LeighAS, 2022-07-06:
[image]

LeighAS, 2022-07-06:
the above build step is where it has appeared to stall but it could be still going. I am off abroad for a few days now. i will leave it running and see whether its finished when i return

solmoller, 2022-07-07:
It's been a long time! Maybe I need to look into a simpler distribution

For now

Uninstall SQLite and try

sudo apt-get install sqlite3 libsqlite3-dev

LeighAS, 2022-07-11:
used
sudo apt install libdbd-sqlite3-perl
and now appear to have a working setup, thanks for all the help, not really a linux guru yet much less a perl pro.

closed 2022-07-11 by LeighAS