solmoller/eversolar-monitor #63: MQTT ha discovery is incomplete with multiple inverters due to nonunique ids

LeighAS, 2022-08-31:
[image]
as seen in mqtt explorer, the second inverter has been given the unique ID of the first one this is most likely the reason that the home assistant instance can not auto discover the second inverter as a separate device.

I am raising this as i'm not a perl or mqtt pro and would rather not make erroneous changes to the perl script. I will have a go if there is no help forthcoming. I hope I am being of some use here.

LeighAS, 2022-09-01:
[image]
finally fixed it. basically its seems its a problem of scope.

 my $mqtt_serial = $inverters{$inverter}{'serial'};
is being declared and set outside of the  sub ha_disc_config() which when I look at mqtt explorer is only setting one of the inverter serials for all of the mqtt messages sent with auto discovery.
the non HA auto discovery messages are however correct so to fix for my rpi instance I brought the initialising the variable inside to a local one
[image]

This was cause I didn't want to mess about which the non home assistant auto discovery serial which appeared to be coming out ok.

I wouldn't say this is perfect as a fix but certainly now home assistant now sees both devices and will probably auto find the 3rd inverter when I fix it and add it in.

open to suggestions if people think there is a better solution?

nagydavid, 2022-09-08:
Can you please submit your changes as a PR, and I can review it and see if it needs any adjustment

solmoller, 2024-02-11:
I can confirm that this change works, here is the modified code:
`               sub ha_disc_config {
                       my $mqtt_serial_HA = $inverters{$inverter}{'serial'};
                       my %config_data = (
                           device => {
                               identifiers => [
                                   $mqtt_serial_HA,
                                   ],
                               manufacturer => "Eversolar",
                               model => $mqtt_inverter_model,
                               name => "Solar Inverter"
                           },
                           state_topic => "$mqtt_topic_prefix/$mqtt_serial_HA/$_[0]",
                           unique_id => "$mqtt_serial_HA\_$_[0]",
                           state_class => "measurement",
                       );`

I now finally have both inverters in HA

solmoller, 2024-02-11:
Added to code

closed 2024-02-11 by solmoller