solmoller/eversolar-monitor #65: Anything possible when no RS485 communications module?

ginoputrino, 2023-02-10:
Hi, 
I have recently moved into a house with an eversolar TL2000, and would like to get this software up and monitoring, but the inverter does not have the RS485 RJ45 comms module, only the socket where it should go: [Imgur Photo here](https://imgur.com/a/fvh6Msn).

Does anyone know where I could buy an old comms module, or failing that, help me to reverse engineer the board so I could build my own?

Even a photo of someone's communications board would help.

Thanks!

wotid, 2023-02-10:
I don't know where you can source one, but I can show you a picture.

Steve.
[image]

solmoller, 2023-02-10:
That is a ten year old inverter. I'd recommend buying an entire used unit.
That way you have a Comms board and a spare unit.

Good luck!

Henrik

ginoputrino, 2023-02-12:
> I don't know where you can source one, but I can show you a picture.
> 
> Steve. [image]

Thanks Steve, looks like a transceiver chip and a few voltage levellers. Might be possible to put something together if I can't find the part.

ginoputrino, 2023-02-12:
> 

Hi Henrik, If I can find a secondhand one, I will totally do that. However I'm not sure how common these units were in my country (Australia).
Otherwise, if I'm not successful finding parts, I will post a kicad file of what I put together. If the 7V line can provide enough current, I might even add a 5V regulator so that I can add a rpi zeroW inside the unit.

wotid, 2023-02-12:
Where in Australia are you? I can mail you one of mine. Only thing is I can't get to the post office until next Saturday at the earliest.

ginoputrino, 2023-02-12:
Steve, that would be awesome. I live in Sydney. How can I send you a private message?

wotid, 2023-02-12:
Send me an email wotid1@gmail.com when I get it I'll just delete this message.

ginoputrino, 2023-02-12:
Ok, done, feel free to delete your message.

ginoputrino, 2023-02-27:
Hi Guys, wanted to thank the very excellent @wotid for sending me his spare RS485 card.
Since I also didn't have the plate with a cable gland and my inverter is in a location that can sometimes see some weather, I've chosen to make a very small package that could live in the inverter so that everything kept it's water tight seal. 

I'm using a raspberry pi zero W, a buck converter to power it by running off the 7V line, and a very small WITMOTION RS485 to USB adapter, which I can confirm works very well.

Since the 7V line shuts down at night, I'm running the rpi with the SD card in read only mode which is a really great feature for this kind of thing, and sending data out via MQTT to my home assistant server.

PV output here: https://pvoutput.org/list.jsp?userid=111270

[Pic here.](https://imgur.com/a/pnj4Qyw)

Thanks again!

wotid, 2023-02-28:
Good job Gino and good to see the data going to PVOutput

closed 2023-02-28 by ginoputrino