solmoller/eversolar-monitor #66: Active power control

solmoller, 2023-08-16:
We should add power control to the system, as negative prices become normal..

Here is the command to set inverter at 90%
Header 
```
Header         Source Address Destination Address Control Code Function Code Data length Data0 Data1 Checksum
0xAA             2 Bytes               2 Bytes                     0x13                   0x20                 0x02          90      30        2 Bytes
```
Data1 is adjust time in  seconds

Integration to HA the other way may pose an interesting problem

henrykuijpers, 2024-01-27:
@solmoller  Is there any specification sheet and/or documentation about the inverter? I sent an e-mail to SMA / Zeversolar, but they told me there is no specification of the protocol. How did you find out about this (nowhere documented?) command?

henrykuijpers, 2024-01-27:
Also, integration to HA could be realized pretty easily, no? A slider could be added that defaults to 100%, that can be controlled. If one controls the slider, the Perl script should be able to catch that configuration (read mqtt for example?) and interact with the inverter to set the desired percentage.

I am very curious to know what more configurations / actions can take place with this inverter. It seems to have a lot of functionality, that is not visible at first sight. I thought it only had a button to scroll through a menu that contains very limited information. There is definitely, in the menu of the inverter, no option to set that power percentage, I think?

solmoller, 2024-02-11:
Hi, send me a PM, and I'll find the factory documentation for you

henrykuijpers, 2024-02-11:
@solmoller how can I send you a pm? I don't think GitHub has a pm functionality?

solmoller, 2024-02-11:
Interesting, I thought that was possible, ill delete this when you reply Henrik  (a7t) lillefamilie dot dk

henrykuijpers, 2024-02-11:
Got it!

solmoller, 2024-02-11:
Got what? ;-)

tobiasnorbo, 2025-03-30:
Hi all,

I need volunteers to test and give feedback/help!

I used many hours last year trying to figure out a way to do this, every single attempt failed. I have tried again now that prices are decreasing due to more sun in the grid. And I think I have succeeded, but I have removed a lot of functionality in the script in order to focus on only this and debugging the script.

What remains is:

- all the communication with the inverter(s) - but only over serial.
- everything MQTT related to publish to HA - since I am using this as my database - this also mean that I have removed the local database.

I will not write all the stuff down that has been removed - I am sure you can figure that out from above description if you know the script.

I have added:

- MQTT listening feature to listen for messages such as: 'Limit power 5' for limiting power to 5%. I have made a lower limit of 5% since I am afraid that 0% would shut down the inverter without being able to get it back up. Later I found out that after 10 minutes it will go back to 100% (according to the protocol) if no new limit power messages is published.
- I have ensure that the power limit message is parsed using the execution function according to the protocol.
- I am parsing the limit power function every 9 minutes to avoid going back to 100% according to the protocol this will happen after 10 minutes.
- On startup I have decided to parse power limit to 100% since the production in the morning is low and most likely will not be higher than the demand in the household for the (maximum) first hour until the first limit power message could potentially be published due to negative (or low) prices.

I have changed:

- The main loop: I had huge issues with the main loop that blocked the MQTT listening feature, so I placed the main loop in a sub routine, which didn't help. Then I changed it from while (42) to an AnyEvent control. This worked and it will now listen to MQTT being published, such as: `mosquitto_pub -h 192.168.X.XXX -p 1883 -u MQTTbroker_name -P MQTT_password -t "eversolar-monitor/command" -m "limit power 5"` - this will be readable in the terminal log (might have to put debugging log at 4).

The issue:

I can see that the correct data is parsed to the inverter, and it also receives the data, but it doesn't limit the power. Supposedly it should slowly limit the power and then be steady around or close to the limit until the 10 minutes has passed unless the power limit message is parsed again, which I have ensured it is. But for my inverter I am not seeing any power limiting. So here is where I really need your help. I would very much like someone to test out my script. So please comment here if you would like to test it out, then I will happily send it to you - for instance by email. I am not ready share it with a wide audience just yet.

Looking forward to hearing from some of you.

NB my inverter is a Zeversolar ZL2000S - you are welcome with any inverter, that has used the eversolar monitor script successfull before.

Thanks!

solmoller, 2025-03-30:
Did you try forcing the limitation without any other script running?

tobiasnorbo, 2025-04-01:
I believe I have found the issue. On the inverter I was supposed to set it to active power function. Press and hold the grey button, and choose active power function on. Now it works. Yeah!

[image]
[image]

tobiasnorbo, 2025-04-02:
Today the script was put to the real life scenario test, and it seems that the active power function was disabled after shutting down during the night. I will need to see if it is possible to set this function via the script or if it is possible to have it on always. Any input on this could be very helpful.

tobiasnorbo, 2025-04-07:
I have succeeded.

If you send a retained MQTT messsage to limit the power, then it works also in the morning. A work-around I haven't tested if is actually needed is setting the limit power (when no limit is needed) to 99% to ensure that the function runs continuously. 

Is anyone interested in the script? What should I do from here to the benefit of the community?

henrykuijpers, 2025-04-07:
@tobiasnorbo create a pull request for @solmoller to review. :)

tobiasnorbo, 2025-04-07:
It seems that I do not have permission to do so - can you help with this @solmoller ?

henrykuijpers, 2025-04-07:
@tobiasnorbo you can make a fork. There, you will have permission to do commits etc. 
I recommend creating a separate branch in your fork. 

Then, when you're done, you can make a pull request here. :)

tobiasnorbo, 2025-04-07:
Thank you! This was my first time trying this - but now there is a pull request uploaded.

Have fun :)

tobiasnorbo, 2025-05-10:
Did anyone try my script? 
I'm just hoping that I'm not the only one taking advantage of the inverters possibilities:)

henrykuijpers, 2025-05-11:
@tobiasnorbo i would have loved to use your code, but I recently upgraded my solar panels and switched to a Growatt inverter. 

I've been very charmed about my old Zeversolar TL-3000 though. It had quite the capabilities for that time.

jiquera, 2025-07-19:
Hi I have a TL3680 here, which i;ve been playing and profiling for a while now. There are a few peculiarities as there seem to be several undocumented func-codes for read(0x11). However, it also doesnt do anything when sending 0x13 0x20 with data and adjust time. It gives an ack (0x06) if it's in the correct range but then nothing happens... any ideas?

@tobiasnorbo Im happy to try your script if I can get my inverter to do something with it

jiquera, 2025-07-28:
I experimented with dozens of commands with or without data, but still havent figured out how to set the inverter to active power control mode.... nobody in my surrounding has a ZeverCom to tap into so @solmoller you are my last hope... :(

jiquera, 2025-08-26:
Thanks for the effort but this is a dump from the script, which I already know exactly what it does (even your version). I need a dump from the communication between a ComBox and an Inverter (raw bytes)... but I've kind lost hope for getting that one :(

tobiasnorbo, 2025-08-26:
Debug is set to 4, but I cannot check it anylonger since it is too long ago the sun went down. I will check again one of the coming days and see if there is more info.