solmoller/eversolar-monitor #70: Integration to ES32?

tobiasnorbo, 2024-04-14:
To keep the price down I was wondering if it would be possible to integrate the software into running on an ESP32 instead of a RasberryPi?
In that case I would be happy to test it out. I do not have enough knoweldge software wise to do it myself. I might try anyway, but I will probably not be successful. I have a Zeversolar 2000s without combox, so I don't know what kind of data I can get out of the two ethernet ports anyway.

solmoller, 2024-04-14:
Perl on ESP32 seems a bit far fetched. You can probably buy an RPI 2 or 3
from someone's drawer, and have a much better experience.

And it'll protect your sanity better.

BR
Henrik

nagydavid, 2024-04-14:
I am running it on a raspberry pi zero version 1 for years without any hiccup.

tobiasnorbo, 2024-04-14:
Thanks, guys. Great feedback - will save me a lot of time :) 

Will look into a RPI instead.

closed 2024-05-02 by tobiasnorbo