solmoller/eversolar-monitor #71: RS485 to USB converter

tobiasnorbo, 2024-05-01:
Hi,

I'm sorry, but I don't know, where else to ask this. Would this converter be sufficient? And in that case, how would you connect the wires of an RS485 cable?

https://raspberrypi.dk/produkt/industriel-usb-til-rs485-bidirectional-converter/

solmoller, 2024-05-01:
That looks promising. Please let me know if it works

In addition, if you use a new rpi with current OS, i'd like to hear if the
installation procedure needs updating. The old sd card image is outdated.

I run on an updated RPI4, but the OS is highly modified for other purposes

BR Henrik

solmoller, 2024-05-01:
> I will let you know. I was just concerned, that there is only A+, B- and
> GND on this converter. Which wires would I connect to these?
>
> I have bought a RPI2 B used, so I am not able to provide input on that
> part just yet :)

That is in the WIKI

tobiasnorbo, 2024-05-02:
I did something right and probably a lot wrong. I was not sure what to do with the USB setting in the ini file. Does it matter which port it goes into and what this setting is?Before restart I did get some numbers, but they were wrong.

tobiasnorbo, 2024-05-02:
I thought it added the screen shots. Here they are. 
[image]
[image]
[image]
[image]

solmoller, 2024-05-02:
> I figured it out. I should have used "serial" and not "eth2ser".
>
> It works great now. But if you read this. Could you point me in the
> direction of how to get this into HA. And is there a way to shut the
> inverter down or stop export to grid?

Great!

Yes, latest version can do HA, and we have an open issue on limiting the
inverter

Good news: you can throttle the thing!

tobiasnorbo, 2024-05-02:
That sounds great! I am looking forward to that addition, so we can help the grid, when there is too much solar power on it.

Do you have a link to some documentation on getting the data into HA? The Danish forum that the wiki links to is not available anymore.

solmoller, 2024-05-02:
I just updated the Wiki, it's MQTT that does the trick

And as for throttling, I have other projects, so I doubt I'll find time. Maybe you can lend a hand?

I am also interested in hearing what you went through to install on modern RPI

BR

Henrik

solmoller, 2024-05-02:
Throttling is #66  - so you used the image in an RPI2 with success?

tobiasnorbo, 2024-05-02:
Thanks a lot. I will look into the wiki and get HA up and running.

I would be happy to contribute, but I'm not sure I will get it to work, since my coding skills are limited.

And yes, I succeeded in using the image for the RPI2, and to complete the hardware I successfully used:

The rs485 converter in question for this 'issue', and it worked perfectly. See link below.
http://www.waveshare.com/wiki/USB_TO_RS485

Other hardware used was:
RaspberryPi 2B from 2015
RasPiKey 16gb, I was concerned about the lifetime of the SD cards, so chose this eMMC module version of a SD card: https://www.uugear.com/doc/RasPiKey_UserManual.pdf

Thanks a lot for all the help

closed 2024-05-02 by tobiasnorbo

tobiasnorbo, 2024-05-03:
Just a small update.

I am not able to install musquitto on the Raspbberry wheezy version, that is part of the img file. So I will need to backup the files and upgrade to a newer OS.

I will therefore suggest that the wiki explains, that it is suggested to run on a newer RPI and not use the img file if you want to use MQTT for HA integration.

Right now I run HA on a RPI5, and the eversolar script on RPI2. But I am considering to execute both on the RPI5, now that I am running into these problems.

solmoller, 2024-05-04:
Thanks! I have updated the Wiki 

I did a crash installation on my RPI4, simply by adding modules as the perl script failed. 
I did not write down the work though..