solmoller/eversolar-monitor #74: Energy tab in Home Assistant

solmoller, 2024-06-23:
My inverters still don't show up in the energy tab in Home Assistant

Logfile:
2024-06-23 09:40:55.728 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.pv_last_365_days_production_2 (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using state class 'measurement' which is impossible considering device class ('energy') it is using; expected None or one of 'total', 'total_increasing'; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22

nagydavid, 2024-06-23:
Hi @solmoller 
You should use sensor.solar_inverter_pv_total_energy_production. The last 365 is not monoton increasing sensor so it cannot be used.
[image]

solmoller, 2024-06-23:
Checked, we don't have last_reset defined.